ros2/kilted_tutorial_party #405: ros2cli network tests

tfoote, 2025-05-01:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Setup
- Rmw: Zenoh
- BuildType: Binary
- Chip: Arm64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros2/ros2cli)

## Checks
- [ ] **Node discovery**

     With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.
     
- [ ] **Topic discovery**

     With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.
     
- [ ] **Service discovery**

     With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.
     
- [ ] **Parameter discovery**

     With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Barry-Xu-2018, 2025-05-08:
PLATFORM INFORMATION
Hardware: Raspberry Pi 4 Model B
System : Ubuntu 24.04.2
processor : aarch64

RMW MIDDLEWARE
middleware name : rmw_zenoh_cpp

ROS 2 INFORMATION
distribution name : kilted
distribution type : ros2
distribution status : pre-release
release platforms : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}

### Test result 

All tests pass

### Test Steps

Prepare ROS 2 Kilted environment (Refer to https://docs.ros.org/en/kilted/Installation/Ubuntu-Install-Debs.html)

Start the Zenoh router in terminal 1
```bash
$ source /opt/ros/kilted/setup.bash
$ ros2 run rmw_zenoh_cpp rmw_zenohd
```

#### Node discovery

Run demo application in a new terminal
```bash
$ source /opt/ros/kilted/setup.bash
$ export RMW_IMPLEMENTATION=rmw_zenoh_cpp
$ ros2 run demo_nodes_cpp talker
[INFO] [1746684671.093498493] [talker]: Publishing: 'Hello World: 1'
[INFO] [1746684672.093392647] [talker]: Publishing: 'Hello World: 2'
[INFO] [1746684673.093383290] [talker]: Publishing: 'Hello World: 3'
...
```

Run `node list`
```bash
$ source /opt/ros/kilted/setup.bash
$ export RMW_IMPLEMENTATION=rmw_zenoh_cpp
$ ros2 node list
/talker
```

Stop demo application and run `node list` again 
```bash
$ ros2 node list
```

#### Topic discovery

Run demo application in a new terminal
```bash
$ source /opt/ros/kilted/setup.bash
$ export RMW_IMPLEMENTATION=rmw_zenoh_cpp
$ ros2 run demo_nodes_cpp talker
[INFO] [1746684671.093498493] [talker]: Publishing: 'Hello World: 1'
[INFO] [1746684672.093392647] [talker]: Publishing: 'Hello World: 2'
[INFO] [1746684673.093383290] [talker]: Publishing: 'Hello World: 3'
...
```

Run `topic list`
```bash
$ source /opt/ros/kilted/setup.bash
$ export RMW_IMPLEMENTATION=rmw_zenoh_cpp
$ ros2 topic list
/chatter
/parameter_events
/rosout
```

Stop demo application and run `topic list` again 
```bash
$ ros2 topic list
```

#### Service discovery

Run demo application in a new terminal
```bash
$ source /opt/ros/kilted/setup.bash
$ export RMW_IMPLEMENTATION=rmw_zenoh_cpp
$ ros2 run demo_nodes_cpp add_two_ints_server
```

Run `service list`
```bash
$ source /opt/ros/kilted/setup.bash
$ export RMW_IMPLEMENTATION=rmw_zenoh_cpp
$ ros2 service list
/add_two_ints
/add_two_ints_server/describe_parameters
/add_two_ints_server/get_parameter_types
/add_two_ints_server/get_parameters
/add_two_ints_server/get_type_description
/add_two_ints_server/list_parameters
/add_two_ints_server/set_parameters
/add_two_ints_server/set_parameters_atomically
```

Stop demo application and run `service list` again 
```bash
$ ros2 service list
```

#### Parameter discovery

Run demo application in a new terminal
```bash
$ source /opt/ros/kilted/setup.bash
$ export RMW_IMPLEMENTATION=rmw_zenoh_cpp
$ ros2 run demo_nodes_cpp parameter_blackboard
[INFO] [1746686085.536882975] [parameter_blackboard]: Parameter blackboard node named '/parameter_blackboard' ready, and serving '6' parameters already!
```

Run `param list`
```bash
$ source /opt/ros/kilted/setup.bash
$ export RMW_IMPLEMENTATION=rmw_zenoh_cpp
$ ros2 param list
/parameter_blackboard:
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  start_type_description_service
  use_sim_time
```

Stop demo application and run `param list` again 
```bash
$ ros2 param list
```

ahcorde, 2025-05-08:
Thank you for testing

closed 2025-05-08 by ahcorde