ros2/kilted_tutorial_party #407: ros2cli network tests

tfoote, 2025-05-01:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Setup
- Rmw: Zenoh
- BuildType: Debian
- Chip: Arm64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros2/ros2cli)

## Checks
- [ ] **Node discovery**

     With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.
     
- [ ] **Topic discovery**

     With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.
     
- [ ] **Service discovery**

     With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.
     
- [ ] **Parameter discovery**

     With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

christophebedard, 2025-05-01:
Thanks for testing! Can you provide the output of the commands you ran?

pradyum, 2025-05-01:
**1st Instance:**

.bashrc modification:

```
source /opt/ros/kilted/setup.bash
export RMW_IMPLEMENTATION=rmw_zenoh_cpp
export ZENOH_CONFIG_OVERRIDE='mode="client";connect/endpoints=["tcp/10.0.1.230:7447"]'
```

terminal-1 : 

```
~$ ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```

**2nd Instance:**

.bashrc modification:

```
source /opt/ros/kilted/setup.bash
export RMW_IMPLEMENTATION=rmw_zenoh_cpp
```

terminal-1 : 

```
~$ ros2 run rmw_zenoh_cpp rmw_zenohd 
2025-05-01T22:46:05.068211Z  INFO ThreadId(02) zenoh::net::runtime: Using ZID: 80656049394d24fa9c646da0dcb29072
2025-05-01T22:46:05.078180Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/[fdef:6def:9c40:0:edac:7aec:3d8a:471b]:7447
2025-05-01T22:46:05.078201Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/[fdef:6def:9c40:0:28dd:b1ff:fe59:97a6]:7447
2025-05-01T22:46:05.078236Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/[fe80::28dd:b1ff:fe59:97a6]:7447
2025-05-01T22:46:05.078239Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/10.0.1.230:7447
Started Zenoh router with id 80656049394d24fa9c646da0dcb29072
```

terminal-2 : 

- [x] Node discovery

```
~$ ros2 node list
/dummy_joint_states
/dummy_laser
/dummy_map_server
/robot_state_publisher
```

- [x] Topic discovery

```
~$ ros2 topic list
/joint_states
/map
/parameter_events
/robot_description
/rosout
/scan
/tf
/tf_static
```

- [x] Service discovery

```
~$ ros2 service list
/dummy_joint_states/describe_parameters
/dummy_joint_states/get_parameter_types
/dummy_joint_states/get_parameters
/dummy_joint_states/get_type_description
/dummy_joint_states/list_parameters
/dummy_joint_states/set_parameters
/dummy_joint_states/set_parameters_atomically
/dummy_laser/describe_parameters
/dummy_laser/get_parameter_types
/dummy_laser/get_parameters
/dummy_laser/get_type_description
/dummy_laser/list_parameters
/dummy_laser/set_parameters
/dummy_laser/set_parameters_atomically
/dummy_map_server/describe_parameters
/dummy_map_server/get_parameter_types
/dummy_map_server/get_parameters
/dummy_map_server/get_type_description
/dummy_map_server/list_parameters
/dummy_map_server/set_parameters
/dummy_map_server/set_parameters_atomically
/robot_state_publisher/describe_parameters
/robot_state_publisher/get_parameter_types
/robot_state_publisher/get_parameters
/robot_state_publisher/get_type_description
/robot_state_publisher/list_parameters
/robot_state_publisher/set_parameters
/robot_state_publisher/set_parameters_atomically
```

- [x] Parameter discovery

```
~$ ros2 param list
/dummy_joint_states:
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  start_type_description_service
  use_sim_time
/dummy_laser:
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  start_type_description_service
  use_sim_time
/dummy_map_server:
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  start_type_description_service
  use_sim_time
/robot_state_publisher:
  frame_prefix
  ignore_timestamp
  publish_frequency
  qos_overrides./joint_states.subscription.depth
  qos_overrides./joint_states.subscription.history
  qos_overrides./joint_states.subscription.reliability
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  qos_overrides./tf.publisher.depth
  qos_overrides./tf.publisher.durability
  qos_overrides./tf.publisher.history
  qos_overrides./tf.publisher.reliability
  qos_overrides./tf_static.publisher.depth
  qos_overrides./tf_static.publisher.history
  qos_overrides./tf_static.publisher.reliability
  robot_description
  start_type_description_service
  use_sim_time
```

christophebedard, 2025-05-01:
Did you also check that the `ros2 ... list` output is empty after stopping the nodes?

pradyum, 2025-05-02:
Yes, ros2 ... list of node, service and param are empty. The ```ros2 topic list``` outputs

```
~$ ros2 topic list 
/parameter_events
/rosout
```

christophebedard, 2025-05-02:
Great, thanks for the info!

closed 2025-05-02 by christophebedard