ros2/kilted_tutorial_party #408: ros2cli network tests

tfoote, 2025-05-01:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Setup
- Rmw: Zenoh
- BuildType: Source
- Chip: Amd64
- Os: Windows

## Links
- [Repo URL](https://github.com/ros2/ros2cli)

## Checks
- [ ] **Node discovery**

     With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.
     
- [ ] **Topic discovery**

     With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.
     
- [ ] **Service discovery**

     With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.
     
- [ ] **Parameter discovery**

     With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

HenryLeC, 2025-05-08:
- [x]  **Node discovery**
With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.

- [x]  **Topic discovery**
With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.

- [x]  **Service discovery**
With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.

- [x]  **Parameter discovery**
With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

```
(pixi_ros2_rolling) PS C:\dev> ros2 node list
/talker
(pixi_ros2_rolling) PS C:\dev> ros2 node list
(pixi_ros2_rolling) PS C:\dev> ros2 topic list
/chatter
/parameter_events
/rosout
(pixi_ros2_rolling) PS C:\dev> ros2 topic list
/parameter_events
/rosout
(pixi_ros2_rolling) PS C:\dev> ros2 service list
/talker/describe_parameters
/talker/get_parameter_types
/talker/get_parameters
/talker/get_type_description
/talker/list_parameters
/talker/set_parameters
/talker/set_parameters_atomically
(pixi_ros2_rolling) PS C:\dev> ros2 service list
(pixi_ros2_rolling) PS C:\dev> ros2 param list
/talker:
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  start_type_description_service
  use_sim_time
(pixi_ros2_rolling) PS C:\dev> ros2 param list
(pixi_ros2_rolling) PS C:\dev>
```

ahcorde, 2025-05-13:
Thank you for testing

closed 2025-05-13 by ahcorde