ros2/kilted_tutorial_party #41: Check ROS 2 doctor CLI

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `-r`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 doctor -r
       ```
  
    **Expect**
    
    1.
       A large report is generated with several sections.
    
  </details>
- [ ] **Check `-rf`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 doctor -rf
       ```
  
    **Expect**
    
    1.
       A lot of checks and a summary of the checks is given (e.g., 'All 5 checks passed').
    
  </details>
- [ ] **Check `-iw`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 doctor -iw
       ```
  
    **Expect**
    
    1.
       A lot of checks and a summary of the checks is given (e.g., '2/5 checks passed').
    
  </details>
- [ ] **Check hello**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 doctor hello
       ```
  
    **Expect**
    
    1.
       You'll see repeated messages that look somewhat like the following.
       ```bash
       # stdout in terminal 1
       MULTIMACHINE COMMUNICATION SUMMARY
       Topic: /canyouhearme, Published Msg Count: 11
       Subscribed from:
                       Hostname             Msg Count /1.0s
       Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 11
       Received from:
                       Hostname             Msg Count /1.0s
       ------------------------------------------------------------
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-02:
It works for me. I have hidden the network configurations from below (IPs) etc. But it was there in the output

```
root@cluttered:~# source /opt/ros/kilted/setup.bash
root@cluttered:~# ./scripts/ros2_doctor_cli.sh 
==========================uname -m
x86_64
==========================middleware
middleware name    : rmw_fastrtps_cpp
==========================ros2 doctor -r

   NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : UP,LOOPBACK,RUNNING
mtu          : 65536
inet         : 192.168.29.137
inet4        : ['192.168.29.137']
ether        : f4:7b:09:6f:5f:81
inet6        : ['2405:201:d000:d170:b306:a034:1275:5935', '2405:201:d000:d170:a1be:4501:9190:2bc9', 'fe80::a901:2c0:950f:8242%wlp0s20f3']
netmask      : 255.255.255.0
device       : wlp0s20f3
flags        : UP,BROADCAST,RUNNING,MULTICAST
mtu          : 1500
broadcast    : 192.168.29.255
inet         : 172.17.0.1
inet4        : ['172.17.0.1']
ether        : 02:42:18:c0:5e:cb
inet6        : ['fe80::42:18ff:fec0:5ecb%docker0']
netmask      : 255.255.0.0
device       : docker0
flags        : UP,BROADCAST,MULTICAST
mtu          : 1500
broadcast    : 172.17.255.255
ether        : 08:8f:c3:28:5f:97
device       : enp44s0
flags        : UP,BROADCAST,MULTICAST
mtu          : 1500

   PACKAGE VERSIONS
uncrustify_vendor                         : latest=3.1.0, local=3.1.0
robot_state_publisher                     : latest=3.4.2, local=3.4.2
ament_cmake_pytest                        : latest=2.7.3, local=2.7.3
ament_pep257                              : latest=0.19.2, local=0.19.2
image_geometry                            : latest=4.1.0, local=4.1.0
tlsf                                      : latest=0.10.1, local=0.10.1
rosbag2_compression                       : latest=0.32.0, local=0.32.0
ament_cmake_ros                           : latest=0.14.3, local=0.14.3
rosidl_pycommon                           : latest=4.9.4, local=4.9.4
ament_cmake_cppcheck                      : latest=0.19.2, local=0.19.2
rpyutils                                  : latest=0.6.2, local=0.6.2
geometry2                                 : latest=0.41.0, local=0.41.0
spdlog_vendor                             : latest=1.7.0, local=1.7.0
rqt_service_caller                        : latest=1.4.0, local=1.4.0
rmw_test_fixture                          : latest=0.14.3, local=0.14.3
rqt_srv                                   : latest=1.3.0, local=1.3.0
urdf_parser_plugin                        : latest=2.12.2, local=2.12.2
osrf_pycommon                             : latest=2.1.4, local=2.1.4
demo_nodes_cpp                            : latest=0.36.0, local=0.36.0
actionlib_msgs                            : latest=5.5.0, local=5.5.0
pcl_conversions                           : latest=2.6.3, local=2.6.3
ament_cmake_flake8                        : latest=0.19.2, local=0.19.2
tango_icons_vendor                        : latest=0.4.0, local=0.4.0
rosidl_runtime_py                         : latest=0.14.1, local=0.14.1
rosidl_generator_py                       : latest=0.24.1, local=0.24.1
libyaml_vendor                            : latest=1.7.1, local=1.7.1
ament_lint_cmake                          : latest=0.19.2, local=0.19.2
[... 289 lines cut ...]

==========================ros2 doctor -iw

All 5 checks passed

==========================ros2 hello
MULTIMACHINE COMMUNICATION SUMMARY
Topic: /canyouhearme, Published Msg Count: 10
Subscribed from:
                Hostname             Msg Count /1.0s
Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 10
Received from:
                Hostname             Msg Count /1.0s
------------------------------------------------------------
MULTIMACHINE COMMUNICATION SUMMARY
Topic: /canyouhearme, Published Msg Count: 10
Subscribed from:
                Hostname             Msg Count /1.0s
Multicast Group/Port: 225.0.0.1/49150, Sent Msg Count: 10
Received from:
                Hostname             Msg Count /1.0s
------------------------------------------------------------
^Croot@cluttered:~# 

```

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde