ros2/kilted_tutorial_party #410: ros2cli on a local machine

tfoote, 2025-05-01:
Test the basic functionality of the ros2cli package family

## Setup
- Rmw: Connext
- BuildType: Binary
- Chip: Amd64
- Os: Windows

## Links
- [Repo URL](https://github.com/ros2/ros2cli)

## Checks
- [ ] **Help text**

     Long-form help text should be available when invoking `ros2 --help` as well as `ros2 <command> --help`
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 -h
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 --help
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 topic -h
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 topic --help
       ```
  
    **Expect**
    
    1.
       Long-form help is available.
    
  </details>
- [ ] **Tab completion**

     When using a supported shell (i.e. bash), tab completion should work for both the bare executable to list commands and for listing sub-commands, like `ros2 <tab>` or `ros2 <command> tab`
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Lists the commands
       ```bash
       # User input in terminal 1
       ros2 <tab><tab>
       ```
    1.
       Completes the word `launch`
       ```bash
       # User input in terminal 1
       ros2 lau<tab>
       ```
    1.
       Completes until `demo_nodes_`
       ```bash
       # User input in terminal 1
       ros2 launch dem<tab>
       ```
    1.
       Completes the package `demo_nodes_cpp`
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_c<tab>
       ```
    1.
       Lists launch files in `demo_nodes_cpp`
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp <tab><tab>
       ```
    1.
       Suggests the launch files name `talker_listener`
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp ta<tab>
       ```
    1.
       Lists several launch files that start with `talker_listener`
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener_<tab><tab>
       ```
    1.
       Completes the launch files name `talker_listener_launch.py`
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener_launch.p<tab>
       ```
  
    **Expect**
    
    1.
       You eventually get to the full command
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener_launch.py
       ```
  </details>
- [ ] **Node discovery**

     With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 node list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /listener
       /talker
       ```
  </details>
- [ ] **Topic discovery**

     With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /chatter
       /parameter_events
       /rosout
       ```
  </details>
- [ ] **Service discovery**

     With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp add_two_ints_server
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 service list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /add_two_ints
       /add_two_ints_server/describe_parameters
       /add_two_ints_server/get_parameter_types
       /add_two_ints_server/get_parameters
       /add_two_ints_server/get_type_description
       /add_two_ints_server/list_parameters
       /add_two_ints_server/set_parameters
       /add_two_ints_server/set_parameters_atomically
       ```
  </details>
- [ ] **Parameter discovery**

     With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run turtlesim turtlesim_node
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 param list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /turtlesim:
         background_b
         background_g
         background_r
         holonomic
         qos_overrides./parameter_events.publisher.depth
         qos_overrides./parameter_events.publisher.durability
         qos_overrides./parameter_events.publisher.history
         qos_overrides./parameter_events.publisher.reliability
         start_type_description_service
         use_sim_time
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-05:
I'm working on this

suchetanrs, 2025-05-05:
It works for me.

The scripts used to run this can be found at https://github.com/suchetanrs/kilted_tutorial_party/tree/main/windows_scripts.

Eventhough it works, there is an issue. I had to wait > 30-40 seconds after killing the node for the nodes to disappear from `ros2 node list`. Same with topic, param and service lists.

The script has a wait period after node killing.

Full log output below:
```
**********************************************************************
** Visual Studio 2022 Developer Command Prompt v17.14.0-pre.5.0
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************

C:\Program Files\Microsoft Visual Studio\2022\Preview>call "\Program Files\rti_connext_dds-7.3.0\resource\scripts\rtisetenv_x64Win64VS2017.bat"
###############################################################################

         (c) Copyright, Real-Time Innovations, All rights reserved.

                           RTI Connext DDS 7.3.0

###############################################################################

C:\Program Files\Microsoft Visual Studio\2022\Preview>set CONNEXTDDS_DIR=C:\Program Files\rti_connext_dds-7.3.0

C:\Program Files\Microsoft Visual Studio\2022\Preview>set RTI_LICENSE_FILE="C:\Users\suche\Downloads\rti_license.dat"

C:\Program Files\Microsoft Visual Studio\2022\Preview>cd C:\pixi_ws

C:\pixi_ws>pixi shell

(pixi_ros2_rolling) C:\pixi_ws>call C:\pixi_ws\ros2-windows\local_setup.bat

(pixi_ros2_rolling) C:\pixi_ws>set RMW_IMPLEMENTATION=rmw_connextdds

(pixi_ros2_rolling) C:\pixi_ws>ros2 doctor --report | grep middleware
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 162: UserWarning: Fail to call NetworkReport class functions.
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 146: UserWarning: Report entry point PackageReport fails to load: No module named 'rosdistro'
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 146: UserWarning: Report entry point PlatformReport fails to load: No module named 'rosdistro'
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 146: UserWarning: Report entry point RosdistroReport fails to load: No module named 'rosdistro'
middleware name    : rmw_connextdds

(pixi_ros2_rolling) C:\pixi_ws>cd kilted_tutorial_party\windows_scripts

(pixi_ros2_rolling) C:\pixi_ws\kilted_tutorial_party\windows_scripts>.\run_ros2_cli.cmd

(pixi_ros2_rolling) C:\pixi_ws\kilted_tutorial_party\windows_scripts>echo "============================================Running ros2 help======================================================"
"============================================Running ros2 help======================================================"

(pixi_ros2_rolling) C:\pixi_ws\kilted_tutorial_party\windows_scripts>call ros2_help.cmd

(pixi_ros2_rolling) C:\pixi_ws\kilted_tutorial_party\windows_scripts>echo "===== running ros2 -h===="
"===== running ros2 -h===="

(pixi_ros2_rolling) C:\pixi_ws\kilted_tutorial_party\windows_scripts>ros2 -h
usage: ros2 [-h] [--use-python-default-buffering] Call `ros2 <command> -h` for more detailed usage. ...

ros2 is an extensible command-line tool for ROS 2.

options:
  -h, --help            show this help message and exit
  --use-python-default-buffering
                        Do not force line buffering in stdout and instead use the python default buffering, which
                        might be affected by PYTHONUNBUFFERED/-u and depends on whatever stdout is interactive or not

Commands:
  action     Various action related sub-commands
  bag        Various rosbag related sub-commands
  component  Various component related sub-commands
[... 141 lines cut ...]
"============================================Running param completion==========================================="
========================== starting turtlesim
========================== param list below
RTI Connext DDS Evaluation License issued to None saudintantrimain@gmail.com For non-production use only.
Expires on 18-may-2025
Please contact support@rti.com with any questions or comments.
/turtlesim:
  background_b
  background_g
  background_r
  holonomic
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  start_type_description_service
  use_sim_time
========================== killing turtlesim
========================== param list after kill
RTI Connext DDS Evaluation License issued to None saudintantrimain@gmail.com For non-production use only.
Expires on 18-may-2025
Please contact support@rti.com with any questions or comments.

(pixi_ros2_rolling) C:\pixi_ws\kilted_tutorial_party\windows_scripts>
```