ros2/kilted_tutorial_party #419: ros2cli on a local machine

tfoote, 2025-05-01:
Test the basic functionality of the ros2cli package family

## Setup
- Rmw: Cyclone
- BuildType: Debian
- Chip: Arm64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros2/ros2cli)

## Checks
- [ ] **Help text**

     Long-form help text should be available when invoking `ros2 --help` as well as `ros2 <command> --help`
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 -h
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 --help
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 topic -h
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 topic --help
       ```
  
    **Expect**
    
    1.
       Long-form help is available.
    
  </details>
- [ ] **Tab completion**

     When using a supported shell (i.e. bash), tab completion should work for both the bare executable to list commands and for listing sub-commands, like `ros2 <tab>` or `ros2 <command> tab`
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Lists the commands
       ```bash
       # User input in terminal 1
       ros2 <tab><tab>
       ```
    1.
       Completes the word `launch`
       ```bash
       # User input in terminal 1
       ros2 lau<tab>
       ```
    1.
       Completes until `demo_nodes_`
       ```bash
       # User input in terminal 1
       ros2 launch dem<tab>
       ```
    1.
       Completes the package `demo_nodes_cpp`
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_c<tab>
       ```
    1.
       Lists launch files in `demo_nodes_cpp`
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp <tab><tab>
       ```
    1.
       Suggests the launch files name `talker_listener`
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp ta<tab>
       ```
    1.
       Lists several launch files that start with `talker_listener`
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener_<tab><tab>
       ```
    1.
       Completes the launch files name `talker_listener_launch.py`
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener_launch.p<tab>
       ```
  
    **Expect**
    
    1.
       You eventually get to the full command
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener_launch.py
       ```
  </details>
- [ ] **Node discovery**

     With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 node list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /listener
       /talker
       ```
  </details>
- [ ] **Topic discovery**

     With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /chatter
       /parameter_events
       /rosout
       ```
  </details>
- [ ] **Service discovery**

     With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp add_two_ints_server
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 service list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /add_two_ints
       /add_two_ints_server/describe_parameters
       /add_two_ints_server/get_parameter_types
       /add_two_ints_server/get_parameters
       /add_two_ints_server/get_type_description
       /add_two_ints_server/list_parameters
       /add_two_ints_server/set_parameters
       /add_two_ints_server/set_parameters_atomically
       ```
  </details>
- [ ] **Parameter discovery**

     With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run turtlesim turtlesim_node
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 param list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /turtlesim:
         background_b
         background_g
         background_r
         holonomic
         qos_overrides./parameter_events.publisher.depth
         qos_overrides./parameter_events.publisher.durability
         qos_overrides./parameter_events.publisher.history
         qos_overrides./parameter_events.publisher.reliability
         start_type_description_service
         use_sim_time
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2025-05-01:
PLATFORM INFORMATION
system           : Linux
platform info    : Linux-6.11.0-24-generic-aarch64-with-glibc2.39
release          : 6.11.0-24-generic
processor        : aarch64

   RMW MIDDLEWARE
middleware name    : rmw_cyclonedds_cpp

   ROS 2 INFORMATION
distribution name      : kilted
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}

**RESULT**

The ros2cli is working as expected and the test have passed

pradyum, 2025-05-05:
**The ros2cli are working fine**

**Help text**

```
~$ ros2 -h
usage: ros2 [-h] [--use-python-default-buffering]
            Call `ros2 <command> -h` for more detailed usage. ...

ros2 is an extensible command-line tool for ROS 2.

options:
  -h, --help            show this help message and exit
  --use-python-default-buffering
                        Do not force line buffering in stdout and
                        instead use the python default buffering,
                        which might be affected by
                        PYTHONUNBUFFERED/-u and depends on
                        whatever stdout is interactive or not

Commands:
  action     Various action related sub-commands
  bag        Various rosbag related sub-commands
  component  Various component related sub-commands
  daemon     Various daemon related sub-commands
  doctor     Check ROS setup and other potential issues
  interface  Show information about ROS interfaces
  launch     Run a launch file
  lifecycle  Various lifecycle related sub-commands
  multicast  Various multicast related sub-commands
  node       Various node related sub-commands
  param      Various param related sub-commands
  pkg        Various package related sub-commands
  run        Run a package specific executable
  security   Various security related sub-commands
  service    Various service related sub-commands
  topic      Various topic related sub-commands
  wtf        Use `wtf` as alias to `doctor`

  Call `ros2 <command> -h` for more detailed usage.
```

```
~$ ros2 topic -h
usage: ros2 topic [-h] [--include-hidden-topics]
                  Call `ros2 topic <command> -h` for more
                  detailed usage. ...

Various topic related sub-commands

options:
  -h, --help            show this help message and exit
  --include-hidden-topics
                        Consider hidden topics as well

Commands:
  bw     Display bandwidth used by topic
  delay  Display delay of topic from timestamp in header
  echo   Output messages from a topic
  find   Output a list of available topics of a given type
  hz     Print the average receiving rate to screen
  info   Print information about a topic
  list   Output a list of available topics
  pub    Publish a message to a topic
  type   Print a topic's type

  Call `ros2 topic <command> -h` for more detailed usage.
```

**Tab completion**

Tab completion works fine - `ros2 launch demo_nodes_cpp talker_listener_launch.py`

**Node discovery**

```
~$ ros2 node list 
/listener
/talker
```

**Topic discovery**

```
~$ ros2 topic list
/chatter
/parameter_events
/rosout
```

**Service discovery**

```
~$ ros2 service list 
/add_two_ints
/add_two_ints_server/describe_parameters
/add_two_ints_server/get_parameter_types
/add_two_ints_server/get_parameters
/add_two_ints_server/get_type_description
/add_two_ints_server/list_parameters
/add_two_ints_server/set_parameters
/add_two_ints_server/set_parameters_atomically
```

**Parameter discovery**

```
~$ ros2 param list
/turtlesim:
  background_b
  background_g
  background_r
  holonomic
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  start_type_description_service
  use_sim_time
```

ahcorde, 2025-05-05:
Thank you for testing