ros2/kilted_tutorial_party #43: Check ROS 2 interface CLI

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check list**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 interface list
       ```
  
    **Expect**
    
    1.
       A large list of all messages, services, and actions is given.
    
  </details>
- [ ] **Check package**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 interface package turtlesim
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       turtlesim/srv/TeleportRelative
       turtlesim/srv/Kill
       turtlesim/msg/Pose
       turtlesim/srv/TeleportAbsolute
       turtlesim/srv/SetPen
       turtlesim/action/RotateAbsolute
       turtlesim/srv/Spawn
       turtlesim/msg/Color
       ```
  </details>
- [ ] **Check packages**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 interface packages
       ```
  
    **Expect**
    
    1.
       A list of all of the packages that define messages, services, or actions is given. An example is as follows.
       
    1.
       ```bash
       # stdout in terminal 1
       action_msgs
       action_tutorials_interfaces
       actionlib_msgs
       builtin_interfaces
       ...
       turtlesim
       unique_identifier_msgs
       visualization_msgs
       ```
  </details>
- [ ] **Check proto**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 interface proto std_msgs/msg/Header
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       "stamp:
         sec: 0
         nanosec: 0
       frame_id: ''
       "
       ```
  </details>
- [ ] **Check show**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 interface show std_msgs/msg/Header
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       # Standard metadata for higher-level stamped data types.
       # This is generally used to communicate timestamped data
       # in a particular coordinate frame.
       
       # Two-integer timestamp that is expressed as seconds and nanoseconds.
       builtin_interfaces/Time stamp
         int32 sec
         uint32 nanosec
       
       # Transform frame with which this data is associated.
       string frame_id
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-03:
I'll test this on Windows 11 / amd64 / binary / CycloneDDS.

ShahazadAbdulla, 2025-05-03:
Completed checks for Issue #43 on Windows 11 / amd64 / binary / CycloneDDS.

- `ros2 interface list`: Passed. Returned a large list as expected.
- `ros2 interface packages`: Passed. Returned a list including turtlesim and others.
- `ros2 interface proto std_msgs/msg/Header`: Passed. Output matched expectation.
- `ros2 interface show std_msgs/msg/Header`: Passed. Output matched expectation.
- **`ros2 interface package turtlesim`**: **FAILED.** This command produced no output, whereas it was expected to list interfaces like `turtlesim/srv/Kill`, `turtlesim/msg/Pose`, etc.

All other checks passed successfully, but `ros2 interface package turtlesim` did not list the interfaces for the package.

ahcorde, 2025-05-09:
The command is wrong, this one is the correct: `ros2 interface package turtlesim_msgs`

ahcorde, 2025-05-09:
I updated the test case https://github.com/osrf/ros2_test_cases/pull/1693

ShahazadAbdulla, 2025-05-09:
you were right @ahcorde and I have tested with the corrected command and it works.
```
(pixi_ros2_rolling) C:\pixi_ws>ros2 interface package turtlesim_msgs
turtlesim_msgs/srv/TeleportAbsolute
turtlesim_msgs/srv/Spawn
turtlesim_msgs/msg/Color
turtlesim_msgs/srv/SetPen
turtlesim_msgs/action/RotateAbsolute
turtlesim_msgs/srv/TeleportRelative
turtlesim_msgs/srv/Kill
turtlesim_msgs/msg/Pose
```

ahcorde, 2025-05-09:
Thank you @ShahazadAbdulla for checking again 👍