ros2/kilted_tutorial_party #432: ros2cli on a local machine

tfoote, 2025-05-01:
Test the basic functionality of the ros2cli package family

## Setup
- Rmw: Zenoh
- BuildType: Debian
- Chip: Arm64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros2/ros2cli)

## Checks
- [ ] **Help text**

     Long-form help text should be available when invoking `ros2 --help` as well as `ros2 <command> --help`
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 -h
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 --help
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 topic -h
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 topic --help
       ```
  
    **Expect**
    
    1.
       Long-form help is available.
    
  </details>
- [ ] **Tab completion**

     When using a supported shell (i.e. bash), tab completion should work for both the bare executable to list commands and for listing sub-commands, like `ros2 <tab>` or `ros2 <command> tab`
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Lists the commands
       ```bash
       # User input in terminal 1
       ros2 <tab><tab>
       ```
    1.
       Completes the word `launch`
       ```bash
       # User input in terminal 1
       ros2 lau<tab>
       ```
    1.
       Completes until `demo_nodes_`
       ```bash
       # User input in terminal 1
       ros2 launch dem<tab>
       ```
    1.
       Completes the package `demo_nodes_cpp`
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_c<tab>
       ```
    1.
       Lists launch files in `demo_nodes_cpp`
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp <tab><tab>
       ```
    1.
       Suggests the launch files name `talker_listener`
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp ta<tab>
       ```
    1.
       Lists several launch files that start with `talker_listener`
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener_<tab><tab>
       ```
    1.
       Completes the launch files name `talker_listener_launch.py`
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener_launch.p<tab>
       ```
  
    **Expect**
    
    1.
       You eventually get to the full command
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener_launch.py
       ```
  </details>
- [ ] **Node discovery**

     With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 node list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /listener
       /talker
       ```
  </details>
- [ ] **Topic discovery**

     With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /chatter
       /parameter_events
       /rosout
       ```
  </details>
- [ ] **Service discovery**

     With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_cpp add_two_ints_server
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 service list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /add_two_ints
       /add_two_ints_server/describe_parameters
       /add_two_ints_server/get_parameter_types
       /add_two_ints_server/get_parameters
       /add_two_ints_server/get_type_description
       /add_two_ints_server/list_parameters
       /add_two_ints_server/set_parameters
       /add_two_ints_server/set_parameters_atomically
       ```
  </details>
- [ ] **Parameter discovery**

     With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run turtlesim turtlesim_node
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 param list
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 2
       /turtlesim:
         background_b
         background_g
         background_r
         holonomic
         qos_overrides./parameter_events.publisher.depth
         qos_overrides./parameter_events.publisher.durability
         qos_overrides./parameter_events.publisher.history
         qos_overrides./parameter_events.publisher.reliability
         start_type_description_service
         use_sim_time
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2025-05-01:
PLATFORM INFORMATION
system           : Linux
platform info    : Linux-6.11.0-24-generic-aarch64-with-glibc2.39
release          : 6.11.0-24-generic
processor        : aarch64

   RMW MIDDLEWARE
middleware name    : rmw_zenoh_cpp

   ROS 2 INFORMATION
distribution name      : kilted
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}

**RESULT**

The ros2cli is working as expected.

This test have **passed**

christophebedard, 2025-05-02:
Thanks for testing. Including the commands you ran and their output would be really appreciated, at least for the main commands.

ahcorde, 2025-05-02:
Thank yo for testing

closed 2025-05-02 by ahcorde