ros2/kilted_tutorial_party #436: RQT

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Connext
- BuildType: Binary
- Chip: Amd64
- Os: Windows

## Links
- [Repo URL](https://github.com/ros-visualization/rqt)

## Checks
- [ ] **Check RQT**

     Launch using the `rqt` command.
     
     The rqt window should be generally usable and should be able to populate some of the discovered plugins.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rqt
       ```
    1.
       In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
       
    1.
       With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
       
    1.
       With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
       
    1.
       With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
       
  
    **Expect**
    
    1.
       Rqt runs smoothly and the plugins work as expected.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-05:
Working on this

suchetanrs, 2025-05-05:
All of them work fine for me 

However, RQT complains about not finding a module named `cairo` although the outputs in the GUI are OK.

The full log output is below the images

[image]
[image]
[image]

Full log output below:
```
**********************************************************************
** Visual Studio 2022 Developer Command Prompt v17.14.0-pre.5.0
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************

C:\Program Files\Microsoft Visual Studio\2022\Preview>cd C:\pixi_ws

C:\pixi_ws>pixi shell

(pixi_ros2_rolling) C:\pixi_ws>call C:\pixi_ws\ros2-windows\local_setup.bat
"[rti_connext_dds_cmake_module][warning] RTI Connext DDS environment script not found (\resource\scripts\rtisetenv_x64Win64VS2017.bat). RTI Connext DDS will not be available at runtime, unless you already configured PATH manually."

(pixi_ros2_rolling) C:\pixi_ws>set RMW_IMPLEMENTATION=rmw_connextdds

(pixi_ros2_rolling) C:\pixi_ws>ros2 doctor --report | grep middleware
[ERROR] [1746442533.415017800] [rcl]: Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_connextdds'), with error message 'failed to load shared library 'rmw_connextdds.dll' due to LoadLibrary error: 126, at C:\ci\ws\src\ros2\rcutils\src\shared_library.c:156, at C:\ci\ws\src\ros2\rmw_implementation\rmw_implementation\src\functions.cpp:67', exiting with 1., at C:\ci\ws\src\ros2\rcl\rcl\src\rcl\rmw_implementation_identifier_check.c:149

(pixi_ros2_rolling) C:\pixi_ws>.pixi

(pixi_ros2_rolling) C:\pixi_ws>call "\Program Files\rti_connext_dds-7.3.0\resource\scripts\rtisetenv_x64Win64VS2017.bat"
###############################################################################

         (c) Copyright, Real-Time Innovations, All rights reserved.

                           RTI Connext DDS 7.3.0

###############################################################################

(pixi_ros2_rolling) C:\pixi_ws>set CONNEXTDDS_DIR=C:\Program Files\rti_connext_dds-7.3.0

(pixi_ros2_rolling) C:\pixi_ws>set RTI_LICENSE_FILE="C:\Users\suche\Downloads\rti_license.dat"

(pixi_ros2_rolling) C:\pixi_ws>ros2 doctor --report | grep middleware
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 162: UserWarning: Fail to call NetworkReport class functions.
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 146: UserWarning: Report entry point PackageReport fails to load: No module named 'rosdistro'
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 146: UserWarning: Report entry point PlatformReport fails to load: No module named 'rosdistro'
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 146: UserWarning: Report entry point RosdistroReport fails to load: No module named 'rosdistro'
middleware name    : rmw_connextdds

(pixi_ros2_rolling) C:\pixi_ws>rqt
RosPluginProvider._parse_plugin_xml() plugin file "C:\pixi_ws\ros2-windows\share\rqt_gui_cpp/plugin.xml" in package "rqt_gui_cpp" not found
RosPluginProvider._parse_plugin_xml() plugin file "C:\pixi_ws\ros2-windows\share\rqt_gui_cpp/plugin.xml" in package "rqt_gui_cpp" not found
RTI Connext DDS Evaluation License issued to None suchetan.saravanan@gmail.com For non-production use only.
Expires on 18-may-2025
Please contact support@rti.com with any questions or comments.
QLayout::removeWidget: Cannot remove a null widget.
RosPluginProvider.load(rqt_bag_plugins/BagImagePlugin) exception raised in __builtin__.__import__(rqt_bag_plugins.image_plugin, [ImagePlugin]):
Traceback (most recent call last):
  File "C:\pixi_ws\ros2-windows\Lib\site-packages\rqt_bag_plugins\image_helper.py", line 38, in <module>
    import cairo
ModuleNotFoundError: No module named 'cairo'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\pixi_ws\ros2-windows\Lib\site-packages\rqt_gui\ros_plugin_provider.py", line 80, in load
    module = __builtin__.__import__(
             ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\pixi_ws\ros2-windows\Lib\site-packages\rqt_bag_plugins\image_plugin.py", line 33, in <module>
    from .image_timeline_renderer import ImageTimelineRenderer
  File "C:\pixi_ws\ros2-windows\Lib\site-packages\rqt_bag_plugins\image_timeline_renderer.py", line 50, in <module>
    from rqt_bag_plugins import image_helper
  File "C:\pixi_ws\ros2-windows\Lib\site-packages\rqt_bag_plugins\image_helper.py", line 40, in <module>
    import cairocffi as cairo
ModuleNotFoundError: No module named 'cairocffi'

rqt_bag.TimelineFrame.load_plugins() failed to load plugin "rqt_bag_plugins/BagImagePlugin":
No module named 'cairocffi'
[INFO] [1746442732.395905200] [rqt_bag.BagWidget]: Recording to C:/pixi_ws/2025-05-05-16-28-48.
[WARN] [1746442732.588189400] [rqt_bag.Rosbag2]: get_entry - open for writing only, returning None
[WARN] [1746442732.595864800] [rqt_bag.Rosbag2]: get_entries_in_range - open for writing only, returning None

```

Yadunund, 2025-05-09:
> However, RQT complains about not finding a module named cairo although the outputs in the GUI are OK.

 Do you mind opening a ticket in https://github.com/ros-visualization/rqt_bag to document this? 

Thanks for testing!

closed 2025-05-09 by Yadunund

suchetanrs, 2025-05-17:
> Do you mind opening a ticket in https://github.com/ros-visualization/rqt_bag to document this?

Sure, done!