ros2/kilted_tutorial_party #437: RQT

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Connext
- BuildType: Binary
- Chip: Arm64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros-visualization/rqt)

## Checks
- [ ] **Check RQT**

     Launch using the `rqt` command.
     
     The rqt window should be generally usable and should be able to populate some of the discovered plugins.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rqt
       ```
    1.
       In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
       
    1.
       With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
       
    1.
       With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
       
    1.
       With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
       
  
    **Expect**
    
    1.
       Rqt runs smoothly and the plugins work as expected.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

aehils, 2025-05-01:
Hey! I wanted to pick this up, hoping you can assign to me. Thanks.

aehils, 2025-05-07:
- RQT launched using `rqt` command in terminal. The program successfully plots position[0] of the dummy robot -  presented as a Sine wave.

- With Node Graph plugin, network of active nodes are displayed.

- ROS Bag file saved successfully to local machine.

- With Topic Monitor, messages published to `/joints_states` available to view.

RQT does run smoothly. However, the program did crash after recording the Bag file. I relaunched RQT and it was back up and running again.

aehils, 2025-05-07:
### Snippet from ROS doctor report

PLATFORM INFORMATION
system           : Linux
platform info    : Linux-6.8.0-59-generic-aarch64-with-glibc2.39
release          : 6.8.0-59-generic
processor        : aarch64

   RMW MIDDLEWARE
middleware name    : rmw_connextdds

   ROS 2 INFORMATION
distribution name      : kilted
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}

   TOPIC LIST
topic               : /chatter
publisher count     : 1
subscriber count    : 1
topic               : /joint_states
publisher count     : 1
subscriber count    : 2
topic               : /map
publisher count     : 1
subscriber count    : 0
topic               : /robot_description
publisher count     : 1
subscriber count    : 0
topic               : /scan
publisher count     : 1
subscriber count    : 0
topic               : /tf
publisher count     : 1
subscriber count    : 0
topic               : /tf_static
publisher count     : 1
subscriber count    : 0

Yadunund, 2025-05-09:
> RQT does run smoothly. However, the program did crash after recording the Bag file. I relaunched RQT and it was back up and running again.

Could you open a ticket upstream https://github.com/ros-visualization/rqt_bag with more details on the crash?

Closing this ticket out as it sounds like things worked as expected otherwise.

closed 2025-05-09 by Yadunund

ahcorde, 2025-05-15:
Thank you for testing