ros2/kilted_tutorial_party #439: RQT

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Connext
- BuildType: Debian
- Chip: Arm64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros-visualization/rqt)

## Checks
- [ ] **Check RQT**

     Launch using the `rqt` command.
     
     The rqt window should be generally usable and should be able to populate some of the discovered plugins.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rqt
       ```
    1.
       In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
       
    1.
       With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
       
    1.
       With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
       
    1.
       With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
       
  
    **Expect**
    
    1.
       Rqt runs smoothly and the plugins work as expected.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

slowrunner, 2025-05-05:
I cannot find the requested test rqt-plugins (Visualization->Plot, Introspection->NodeGraph, Logging>Bag, Topic>TopicMonitor)

```
tovli@ros2kiltedubuntu24:~$ dpkg -l | grep rqt
ii  ros-kilted-rqt-action                             2.3.0-2noble.20250502.205615             amd64        rqt_action provides a feature to introspect all available ROS action types.
ii  ros-kilted-rqt-bag                                2.0.2-2noble.20250502.222348             amd64        rqt_bag provides a GUI plugin for displaying and replaying ROS bag files.
ii  ros-kilted-rqt-bag-plugins                        2.0.2-2noble.20250502.224019             amd64        rqt_bag provides a GUI plugin for displaying and replaying ROS bag files.
ii  ros-kilted-rqt-common-plugins                     1.2.0-4noble.20250502.224052             amd64        rqt_common_plugins metapackage provides ROS backend graphical tools suite that can be used on/off of robot runtime.
ii  ros-kilted-rqt-console                            2.3.1-2noble.20250502.202625             amd64        rqt_console provides a GUI plugin for displaying and filtering ROS messages.
ii  ros-kilted-rqt-graph                              1.7.0-2noble.20250502.202841             amd64        rqt_graph provides a GUI plugin for visualizing the ROS computation graph.
ii  ros-kilted-rqt-gui                                1.9.0-2noble.20250502.202241             amd64        rqt_gui provides the main to start an instance of the ROS integrated graphical user interface provided by qt_gui.
ii  ros-kilted-rqt-gui-cpp                            1.9.0-2noble.20250502.203533             amd64        rqt_gui_cpp enables GUI plugins to use the C++ client library for ROS.
ii  ros-kilted-rqt-gui-py                             1.9.0-2noble.20250502.202531             amd64        rqt_gui_py enables GUI plugins to use the Python client library for ROS.
ii  ros-kilted-rqt-image-view                         1.3.0-2noble.20250502.205243             amd64        rqt_image_view provides a GUI plugin for displaying images using image_transport.
ii  ros-kilted-rqt-msg                                1.6.0-2noble.20250502.203102             amd64        A Python GUI plugin for introspecting available ROS message types.
ii  ros-kilted-rqt-plot                               1.6.2-2noble.20250502.202848             amd64        rqt_plot provides a GUI plugin visualizing numeric values in a 2D plot using different plotting backends.
ii  ros-kilted-rqt-publisher                          1.9.0-2noble.20250502.202857             amd64        rqt_publisher provides a GUI plugin for publishing arbitrary messages with fixed or computed field values.
ii  ros-kilted-rqt-py-common                          1.9.0-2noble.20250502.202256             amd64        rqt_py_common provides common functionality for rqt plugins written in Python.
ii  ros-kilted-rqt-py-console                         1.4.0-2noble.20250502.202900             amd64        rqt_py_console is a Python GUI plugin providing an interactive Python console.
ii  ros-kilted-rqt-reconfigure                        1.7.0-2noble.20250502.202833             amd64        This rqt plugin provides a way to view and edit parameters on nodes.
ii  ros-kilted-rqt-service-caller                     1.4.0-2noble.20250502.202949             amd64        rqt_service_caller provides a GUI plugin for calling arbitrary services.
ii  ros-kilted-rqt-shell                              1.3.1-2noble.20250502.203000             amd64        rqt_shell is a Python GUI plugin providing an interactive shell.
ii  ros-kilted-rqt-srv                                1.3.0-2noble.20250502.205626             amd64        A Python GUI plugin for introspecting available ROS service types.
ii  ros-kilted-rqt-topic                              1.8.0-2noble.20250502.205317             amd64        rqt_topic provides a GUI plugin for displaying debug information about ROS topics including publishers, subscribers, publishing rate, and ROS Messages.
tovli@ros2kiltedubuntu24:~$ 

```

akky20, 2025-05-08:
Tested !! 

Terminal 1

[image]

Termianl 2

[image]

[image]

[image]

@slowrunner For me its running fine. Maybe any package issue is there in your setup

closed 2025-05-08 by ahcorde