ros2/kilted_tutorial_party #44: Check ROS 2 interface CLI

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check list**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 interface list
       ```
  
    **Expect**
    
    1.
       A large list of all messages, services, and actions is given.
    
  </details>
- [ ] **Check package**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 interface package turtlesim
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       turtlesim/srv/TeleportRelative
       turtlesim/srv/Kill
       turtlesim/msg/Pose
       turtlesim/srv/TeleportAbsolute
       turtlesim/srv/SetPen
       turtlesim/action/RotateAbsolute
       turtlesim/srv/Spawn
       turtlesim/msg/Color
       ```
  </details>
- [ ] **Check packages**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 interface packages
       ```
  
    **Expect**
    
    1.
       A list of all of the packages that define messages, services, or actions is given. An example is as follows.
       
    1.
       ```bash
       # stdout in terminal 1
       action_msgs
       action_tutorials_interfaces
       actionlib_msgs
       builtin_interfaces
       ...
       turtlesim
       unique_identifier_msgs
       visualization_msgs
       ```
  </details>
- [ ] **Check proto**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 interface proto std_msgs/msg/Header
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       "stamp:
         sec: 0
         nanosec: 0
       frame_id: ''
       "
       ```
  </details>
- [ ] **Check show**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 interface show std_msgs/msg/Header
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       # Standard metadata for higher-level stamped data types.
       # This is generally used to communicate timestamped data
       # in a particular coordinate frame.
       
       # Two-integer timestamp that is expressed as seconds and nanoseconds.
       builtin_interfaces/Time stamp
         int32 sec
         uint32 nanosec
       
       # Transform frame with which this data is associated.
       string frame_id
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-02:
The output below is really long so in short: All of them work!

```
root@cluttered:~# # User input in terminal 1
ros2 interface list
Messages:
    action_msgs/msg/GoalInfo
    action_msgs/msg/GoalStatus
    action_msgs/msg/GoalStatusArray
    actionlib_msgs/msg/GoalID
    actionlib_msgs/msg/GoalStatus
    actionlib_msgs/msg/GoalStatusArray
    builtin_interfaces/msg/Duration
    builtin_interfaces/msg/Time
    diagnostic_msgs/msg/DiagnosticArray
    diagnostic_msgs/msg/DiagnosticStatus
    diagnostic_msgs/msg/KeyValue
    example_interfaces/msg/Bool
    example_interfaces/msg/Byte
    example_interfaces/msg/ByteMultiArray
    example_interfaces/msg/Char
    example_interfaces/msg/Empty
    example_interfaces/msg/Float32
    example_interfaces/msg/Float32MultiArray
    example_interfaces/msg/Float64
    example_interfaces/msg/Float64MultiArray
    example_interfaces/msg/Int16
    example_interfaces/msg/Int16MultiArray
    example_interfaces/msg/Int32
    example_interfaces/msg/Int32MultiArray
    example_interfaces/msg/Int64
    example_interfaces/msg/Int64MultiArray
    example_interfaces/msg/Int8
    example_interfaces/msg/Int8MultiArray
    example_interfaces/msg/MultiArrayDimension
    example_interfaces/msg/MultiArrayLayout
    example_interfaces/msg/String
    example_interfaces/msg/UInt16
    example_interfaces/msg/UInt16MultiArray
    example_interfaces/msg/UInt32
    example_interfaces/msg/UInt32MultiArray
    example_interfaces/msg/UInt64
    example_interfaces/msg/UInt64MultiArray
    example_interfaces/msg/UInt8
    example_interfaces/msg/UInt8MultiArray
    example_interfaces/msg/WString
    geometry_msgs/msg/Accel
    geometry_msgs/msg/AccelStamped
    geometry_msgs/msg/AccelWithCovariance
    geometry_msgs/msg/AccelWithCovarianceStamped
    geometry_msgs/msg/Inertia
    geometry_msgs/msg/InertiaStamped
    geometry_msgs/msg/Point
    geometry_msgs/msg/Point32
    geometry_msgs/msg/PointStamped
    geometry_msgs/msg/Polygon
    geometry_msgs/msg/PolygonInstance
    geometry_msgs/msg/PolygonInstanceStamped
    geometry_msgs/msg/PolygonStamped
    geometry_msgs/msg/Pose
    geometry_msgs/msg/Pose2D
    geometry_msgs/msg/PoseArray
    geometry_msgs/msg/PoseStamped
    geometry_msgs/msg/PoseWithCovariance
    geometry_msgs/msg/PoseWithCovarianceStamped
    geometry_msgs/msg/Quaternion
    geometry_msgs/msg/QuaternionStamped
    geometry_msgs/msg/Transform
    geometry_msgs/msg/TransformStamped
    geometry_msgs/msg/Twist
[... 254 lines cut ...]
  nanosec: 0
frame_id: ''
"
root@cluttered:~# ===================================================================================================
bash: ===================================================================================================: command not found
root@cluttered:~# ===================================================================================================
bash: ===================================================================================================: command not found
root@cluttered:~# ===================================================================================================
bash: ===================================================================================================: command not found
root@cluttered:~# # User input in terminal 1
ros2 interface show std_msgs/msg/Header
# Standard metadata for higher-level stamped data types.
# This is generally used to communicate timestamped data
# in a particular coordinate frame.

# Two-integer timestamp that is expressed as seconds and nanoseconds.
builtin_interfaces/Time stamp
	int32 sec
	uint32 nanosec

# Transform frame with which this data is associated.
string frame_id
root@cluttered:~# 

```

suchetanrs, 2025-05-02:
Seems like I missed a test:

Works with the migrated turtlesim_msgs as mentioned in https://github.com/ros2/kilted_tutorial_party/issues/47

```
root@cluttered:~# source /opt/ros/kilted/setup.bash
root@cluttered:~# export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
root@cluttered:~# ros2 interface package turtlesim_msgs
turtlesim_msgs/action/RotateAbsolute
turtlesim_msgs/srv/TeleportAbsolute
turtlesim_msgs/srv/TeleportRelative
turtlesim_msgs/srv/Kill
turtlesim_msgs/srv/SetPen
turtlesim_msgs/msg/Pose
turtlesim_msgs/srv/Spawn
turtlesim_msgs/msg/Color
root@cluttered:~# 

```

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde