ros2/kilted_tutorial_party #440: RQT

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Connext
- BuildType: Source
- Chip: Amd64
- Os: Windows

## Links
- [Repo URL](https://github.com/ros-visualization/rqt)

## Checks
- [ ] **Check RQT**

     Launch using the `rqt` command.
     
     The rqt window should be generally usable and should be able to populate some of the discovered plugins.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rqt
       ```
    1.
       In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
       
    1.
       With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
       
    1.
       With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
       
    1.
       With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
       
  
    **Expect**
    
    1.
       Rqt runs smoothly and the plugins work as expected.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

mvishnu1404, 2025-05-08:
Hi @tfoote ,
I can work on this. Can you please assign it to me?

HenryLeC, 2025-05-08:
- [x] **Check RQT**

[image]

```
(pixi_ros2_rolling) C:\dev>rqt
RosPluginProvider._parse_plugin_xml() plugin file "C:\dev\kilted\install\share\rqt_gui_cpp/plugin.xml" in package "rqt_gui_cpp" not found
RosPluginProvider._parse_plugin_xml() plugin file "C:\dev\kilted\install\share\rqt_gui_cpp/plugin.xml" in package "rqt_gui_cpp" not found
QLayout::removeWidget: Cannot remove a null widget.
[INFO] [1746739152.389258800] [rqt_bag.BagWidget]: Recording to C:/dev/2025-05-08-17-19-09.
[WARN] [1746739152.766047300] [rqt_bag.Rosbag2]: get_entry - open for writing only, returning None
[WARN] [1746739152.767608000] [rqt_bag.Rosbag2]: get_entry - open for writing only, returning None
[WARN] [1746739152.768822300] [rqt_bag.Rosbag2]: get_entry - open for writing only, returning None
[WARN] [1746739152.770214400] [rqt_bag.Rosbag2]: get_entry - open for writing only, returning None
[WARN] [1746739152.771507600] [rqt_bag.Rosbag2]: get_entry - open for writing only, returning None
[WARN] [1746739152.772811200] [rqt_bag.Rosbag2]: get_entry - open for writing only, returning None
[WARN] [1746739152.774235600] [rqt_bag.Rosbag2]: get_entry - open for writing only, returning None
[WARN] [1746739152.775767200] [rqt_bag.Rosbag2]: get_entry - open for writing only, returning None
[WARN] [1746739152.791191600] [rqt_bag.Rosbag2]: get_entries_in_range - open for writing only, returning None
[WARN] [1746739152.797638400] [rqt_bag.Rosbag2]: get_entries_in_range - open for writing only, returning None
[WARN] [1746739152.806887300] [rqt_bag.Rosbag2]: get_entries_in_range - open for writing only, returning None
[WARN] [1746739152.817177600] [rqt_bag.Rosbag2]: get_entries_in_range - open for writing only, returning None
[WARN] [1746739152.825157900] [rqt_bag.Rosbag2]: get_entries_in_range - open for writing only, returning None
[WARN] [1746739152.834278100] [rqt_bag.Rosbag2]: get_entries_in_range - open for writing only, returning None
[WARN] [1746739152.837485100] [rqt_bag.Rosbag2]: get_entries_in_range - open for writing only, returning None
[WARN] [1746739152.840000000] [rqt_bag.Rosbag2]: get_entries_in_range - open for writing only, returning None
```

Nedded to `pip install cairocffi` to stop rqt error when opening bag widget

zamantousif, 2025-05-14:
There are similar errors when launching plot from inside rqt i.e. rqt -> Plugins -> Visualization -> Plot.

[image]

zamantousif, 2025-05-15:
I tried with RTI 7.3.0 evaluation version as well. I have similar issues as above to report with this version too.
`rqt` does work like above however, launching plot from inside rqt i.e. rqt -> Plugins -> Visualization -> Plot leads to runtime failure and the rqt window exits.
```
(pixi_ros2_rolling) C:\dev\ros2_connextdds>rqt                                                                       
RosPluginProvider._parse_plugin_xml() plugin file "C:\dev\kilted\install\share\rqt_gui_cpp/plugin.xml" in package "rqt_gui_cpp" not found                                                                                                 
RosPluginProvider._parse_plugin_xml() plugin file "C:\dev\kilted\install\share\rqt_gui_cpp/plugin.xml" in package "rqt_gui_cpp" not found     
```
Launching nodes does not work, simple `talker`, `listener` all lead to runtime errors and process exit with failure.
[image]

zamantousif, 2025-05-20:
Tried with RTI 7.3.0 and source workspace on Windows 10 machine.

Sine wave plot of the joint state position

[image]

rqt works smoothly as can be seen in the screenshot
[image]

@tfoote @mjcarroll @ahcorde this ticket could be closed now