ros2/kilted_tutorial_party #443: RQT

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Cyclone
- BuildType: Binary
- Chip: Arm64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros-visualization/rqt)

## Checks
- [ ] **Check RQT**

     Launch using the `rqt` command.
     
     The rqt window should be generally usable and should be able to populate some of the discovered plugins.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rqt
       ```
    1.
       In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
       
    1.
       With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
       
    1.
       With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
       
    1.
       With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
       
  
    **Expect**
    
    1.
       Rqt runs smoothly and the plugins work as expected.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

danielcranston, 2025-05-04:
Following https://docs.ros.org/en/kilted/Installation/RMW-Implementations/DDS-Implementations/Working-with-Eclipse-CycloneDDS.html, I installed the cyclone RMW via apt:

```
$ sudo apt install ros-kilted-rmw-cyclonedds-cpp
$ export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
```

Setup:

```bash
$ dpkg --print-architecture
arm64
$ lsb_release -d | grep Ubuntu
Description:    Ubuntu 24.04.2 LTS
$ echo $RMW_IMPLEMENTATION
rmw_cyclonedds_cpp

$source ~/ros2_kilted_binary_archive/ros2-linux/setup.bash
```

### Checks

- [x] **Check RQT**

[image]

(Running Ubuntu 24.04 over WSL)

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde