ros2/kilted_tutorial_party #447: RQT

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Fastdds
- BuildType: Binary
- Chip: Amd64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros-visualization/rqt)

## Checks
- [ ] **Check RQT**

     Launch using the `rqt` command.
     
     The rqt window should be generally usable and should be able to populate some of the discovered plugins.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rqt
       ```
    1.
       In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
       
    1.
       With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
       
    1.
       With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
       
    1.
       With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
       
  
    **Expect**
    
    1.
       Rqt runs smoothly and the plugins work as expected.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-02:
- [ ] Check RQT

Hi, although I get the data, it is quite glitchy. I ran it on cyclonedds on the same container and it works fine.

https://github.com/user-attachments/assets/d4f57ad4-ced8-48bc-84a7-784479381d01

suchetanrs, 2025-05-02:
I restarted and rebuilt my container. 

The glitchiness with fastdds persists. Cyclonedds, connextdds and zenoh work fine.

JatinVira, 2025-05-13:
Hey,
i tried this out in the latest docker image released and its working smooth as butter for me
here's a screen recording

https://github.com/user-attachments/assets/8f273241-9000-4c2d-a916-3e4627e79b59

Here's my setup
```
   PLATFORM INFORMATION
system           : Linux
platform info    : Linux-5.15.0-139-generic-x86_64-with-glibc2.39
release          : 5.15.0-139-generic
processor        : x86_64

   QOS COMPATIBILITY LIST
topic [type]            : /joint_states [sensor_msgs/msg/JointState]
publisher node          : dummy_joint_states
subscriber node         : robot_state_publisher
compatibility status    : OK
topic [type]            : /parameter_events [rcl_interfaces/msg/ParameterEvent]
publisher node          : _ros2cli_daemon_0_63b64d8f1b8b4464ad736d8747b08379
subscriber node         : robot_state_publisher
compatibility status    : OK
topic [type]            : /parameter_events [rcl_interfaces/msg/ParameterEvent]
publisher node          : dummy_map_server
subscriber node         : robot_state_publisher
compatibility status    : OK
topic [type]            : /parameter_events [rcl_interfaces/msg/ParameterEvent]
publisher node          : robot_state_publisher
subscriber node         : robot_state_publisher
compatibility status    : OK
topic [type]            : /parameter_events [rcl_interfaces/msg/ParameterEvent]
publisher node          : dummy_joint_states
subscriber node         : robot_state_publisher
compatibility status    : OK
topic [type]            : /parameter_events [rcl_interfaces/msg/ParameterEvent]
publisher node          : dummy_laser
subscriber node         : robot_state_publisher
compatibility status    : OK

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : kilted
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}

```

I found all plugins to work as expected.

@ahcorde 
You can assign this issue to me and @suchetanrs  
we can mark this issue completed and close this.

Thankyou

JatinVira, 2025-05-13:
[image]

ahcorde, 2025-05-13:
Thank you for testing

closed 2025-05-13 by ahcorde