ros2/kilted_tutorial_party #449: RQT

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Fastdds
- BuildType: Binary
- Chip: Arm64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros-visualization/rqt)

## Checks
- [ ] **Check RQT**

     Launch using the `rqt` command.
     
     The rqt window should be generally usable and should be able to populate some of the discovered plugins.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rqt
       ```
    1.
       In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
       
    1.
       With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
       
    1.
       With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
       
    1.
       With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
       
  
    **Expect**
    
    1.
       Rqt runs smoothly and the plugins work as expected.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

danielcranston, 2025-05-02:
## Setup

```bash
$ echo $RMW_IMPLEMENTATION
# not set, so default (Fastdds)
$ dpkg --print-architecture
arm64
$ lsb_release -d
No LSB modules are available.
Description:    Ubuntu 24.04.2 LTS
```

All terminals below first source the binary archive:

```bash
$ source ~/ros2_kilted_binary_archive/ros2-linux/setup.bash
```

## Checks

- [x] **Check RQT**

```bash
# Terminal 1
$ ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```

```bash
# Terminal 2
$ rqt
```

All mentioned plugins work as intended.

* 'Plugins > Visualization > Plot'
* 'Plugins > Introspection > Node Graph'
* 'Plugins > Logging > Bag'
* 'Plugins > Topic > Topic Monitor'

[image]

Also as the image indicates, yes I'm using Ubuntu 24.04 over WSL (from a Windows 11 machine). Let me know if that's an issue!

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde