ros2/kilted_tutorial_party #45: Check ROS 2 interface CLI

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check list**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 interface list
       ```
  
    **Expect**
    
    1.
       A large list of all messages, services, and actions is given.
    
  </details>
- [ ] **Check package**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 interface package turtlesim
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       turtlesim/srv/TeleportRelative
       turtlesim/srv/Kill
       turtlesim/msg/Pose
       turtlesim/srv/TeleportAbsolute
       turtlesim/srv/SetPen
       turtlesim/action/RotateAbsolute
       turtlesim/srv/Spawn
       turtlesim/msg/Color
       ```
  </details>
- [ ] **Check packages**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 interface packages
       ```
  
    **Expect**
    
    1.
       A list of all of the packages that define messages, services, or actions is given. An example is as follows.
       
    1.
       ```bash
       # stdout in terminal 1
       action_msgs
       action_tutorials_interfaces
       actionlib_msgs
       builtin_interfaces
       ...
       turtlesim
       unique_identifier_msgs
       visualization_msgs
       ```
  </details>
- [ ] **Check proto**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 interface proto std_msgs/msg/Header
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       "stamp:
         sec: 0
         nanosec: 0
       frame_id: ''
       "
       ```
  </details>
- [ ] **Check show**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 interface show std_msgs/msg/Header
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       # Standard metadata for higher-level stamped data types.
       # This is generally used to communicate timestamped data
       # in a particular coordinate frame.
       
       # Two-integer timestamp that is expressed as seconds and nanoseconds.
       builtin_interfaces/Time stamp
         int32 sec
         uint32 nanosec
       
       # Transform frame with which this data is associated.
       string frame_id
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-04:
I'll test this on Windows 11 / amd64 / binary / Fast DDS

ShahazadAbdulla, 2025-05-04:
Testing Issue #45 on Windows 11 / amd64 / binary / Fast DDS (`rmw_fastrtps_cpp`).

**Results:**
*   `ros2 interface list`: PASSED. Produced a large list of messages, services, and actions.
*   `ros2 interface package turtlesim`: **FAILED.** This command produced no output, instead of the expected list of turtlesim interfaces (e.g., `turtlesim/srv/TeleportRelative`, etc.).
*   `ros2 interface packages`: PASSED. Produced the expected list of packages containing interfaces.
*   `ros2 interface proto std_msgs/msg/Header`: PASSED. Produced the exact expected prototype output.
*   `ros2 interface show std_msgs/msg/Header`: PASSED. Produced the exact expected message definition.

**Conclusion:** Most checks passed, but `ros2 interface package turtlesim` failed to list any interfaces for the package.

mjcarroll, 2025-05-09:
> ros2 interface package turtlesim: FAILED. This command produced no output, instead of the expected list of turtlesim interfaces (e.g., turtlesim/srv/TeleportRelative, etc.).

This is (somewhat) expected. We moved the turtlesim interfaces into https://github.com/ros/ros_tutorials/tree/rolling/turtlesim_msgs, so the check needs to be updated.

closed 2025-05-09 by mjcarroll

ShahazadAbdulla, 2025-05-09:
You were right @mjcarroll and I have tested it and works correctly now.

```
(pixi_ros2_rolling) C:\pixi_ws>ros2 interface package turtlesim_msgs
turtlesim_msgs/srv/TeleportAbsolute
turtlesim_msgs/srv/Spawn
turtlesim_msgs/msg/Color
turtlesim_msgs/srv/SetPen
turtlesim_msgs/action/RotateAbsolute
turtlesim_msgs/srv/TeleportRelative
turtlesim_msgs/srv/Kill
turtlesim_msgs/msg/Pose
```