ros2/kilted_tutorial_party #450: RQT

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Chip: Amd64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros-visualization/rqt)

## Checks
- [ ] **Check RQT**

     Launch using the `rqt` command.
     
     The rqt window should be generally usable and should be able to populate some of the discovered plugins.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rqt
       ```
    1.
       In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
       
    1.
       With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
       
    1.
       With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
       
    1.
       With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
       
  
    **Expect**
    
    1.
       Rqt runs smoothly and the plugins work as expected.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Tenveis, 2025-05-02:
Hi, I'm testing this

Tenveis, 2025-05-03:
-[] Check RQT

Results:

All the rqt windows were open individually.

- []  Lagging
After opening 'Plugins > Visualization > Plot'  the Sine wave graph is lagging. 

[image]

System Monitor status while opening 'Plugins > Visualization > Plot'
[image]

- [X] Works as expected.
[image]

- [X] Works as expected.
[image]

- [X] Works as expected.
[image]

When opening all the windows simultaneously, the system lags and is not smooth.
[image]

my system:
Processor: AMD Ryzen™ 7 5800H with Radeon™ Graphics × 16
Graphics cards: NVIDIA RTX 3050
RAM: 16GB

closed 2025-05-07 by fujitatomoya