ros2/kilted_tutorial_party #451: RQT

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Chip: Arm64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros-visualization/rqt)

## Checks
- [ ] **Check RQT**

     Launch using the `rqt` command.
     
     The rqt window should be generally usable and should be able to populate some of the discovered plugins.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rqt
       ```
    1.
       In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
       
    1.
       With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
       
    1.
       With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
       
    1.
       With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
       
  
    **Expect**
    
    1.
       Rqt runs smoothly and the plugins work as expected.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

slowrunner, 2025-05-05:
@pradyum 

Did you see the plugins?  On my install on Raspberry Pi 4 Ubuntu 24.04.02LTS arm64 TF.svg is missing and causes RQT to abend.

nobso, 2025-05-12:
Everything seems to be working as expected in the following env,
```
RMW Implementation: FASTDDS
BuildType: debian: pre-built debian packages from the apt repository
OS: Ubuntu Noble
Chip: Arm64
```

> In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to /joint_states/position[0]. You should see a Sine wave.

[image]

> With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.

[image]

> With 'Plugins > Logging > Bag' you should be able record a ROS Bag file. 
I could record and save the bag files.

[image]

> With 'Plugins > Topic > Topic Monitor' you should be able to check the /joint_states topic and view messages streaming in.

[image]

ahcorde, 2025-05-12:
Thank you for testing

closed 2025-05-12 by ahcorde