ros2/kilted_tutorial_party #452: RQT

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Fastdds
- BuildType: Source
- Chip: Amd64
- Os: Rhel 9

## Links
- [Repo URL](https://github.com/ros-visualization/rqt)

## Checks
- [ ] **Check RQT**

     Launch using the `rqt` command.
     
     The rqt window should be generally usable and should be able to populate some of the discovered plugins.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rqt
       ```
    1.
       In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
       
    1.
       With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
       
    1.
       With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
       
    1.
       With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
       
  
    **Expect**
    
    1.
       Rqt runs smoothly and the plugins work as expected.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

danielcranston, 2025-05-07:
Setup is identical to https://github.com/ros2/kilted_tutorial_party/issues/296#issuecomment-2860575648

I followed this to allow GUI applications inside the container: https://medium.com/@priyamsanodiya340/running-gui-applications-in-docker-containers-a-step-by-step-guide-335b54472e4b

```
$ xhost +local:docker
$ docker run -it --rm -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix  docker.io/library/almalinux:srcinstall # A tag I comitted manually after finishing #296
[root@1ec5867c3bea ~]# source ros2_kilted/install/setup.bash 
[root@1ec5867c3bea ~]# rqt
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
MESA: error: Failed to query drm device.
glx: failed to create dri3 screen
failed to load driver: iris
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
MESA: error: Failed to query drm device.
glx: failed to create dri3 screen
failed to load driver: iris
```
Some errors pop up, but the RQT window appears without issues.

### Checks

- [ ] ~**Check RQT**~

When adding the first plugin ( 'Plugins > Visualization > Plot' ), the RQT window remains unchanged and I get this error in the terminal runnin RQT;

```
PluginManager._load_plugin() could not load plugin "rqt_plot/Plot":
Traceback (most recent call last):
  File "/root/ros2_kilted/install/qt_gui/lib/python3.9/site-packages/qt_gui/plugin_handler.py", line 102, in load
    self._load()
  File "/root/ros2_kilted/install/qt_gui/lib/python3.9/site-packages/qt_gui/plugin_handler_direct.py", line 55, in _load
    self._plugin = self._plugin_provider.load(self._instance_id.plugin_id, self._context)
  File "/root/ros2_kilted/install/qt_gui/lib/python3.9/site-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/root/ros2_kilted/install/qt_gui/lib/python3.9/site-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/root/ros2_kilted/build/rqt_gui_py/src/rqt_gui_py/ros_py_plugin_provider.py", line 69, in load
    return super(RosPyPluginProvider, self).load(plugin_id, ros_plugin_context)
  File "/root/ros2_kilted/install/qt_gui/lib/python3.9/site-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/root/ros2_kilted/build/rqt_gui/src/rqt_gui/ros_plugin_provider.py", line 107, in load
    return class_ref(plugin_context)
  File "/root/ros2_kilted/build/rqt_plot/src/rqt_plot/plot.py", line 50, in __init__
    self._data_plot = DataPlot(self._widget)
  File "/root/ros2_kilted/build/rqt_plot/src/rqt_plot/data_plot/__init__.py", line 153, in __init__
    raise RuntimeError(
RuntimeError: No usable plot type found. Install at least one of: PyQtGraph, MatPlotLib (at least 1.4.0) or Python-Qwt5.
```

However `dnf list` implies that they exist:
```
# dnf list | grep -iE "matplotl|qtgraph|Qwt5"
python3-matplotlib.x86_64                                                                3.4.3-4.el9                          @epel              
python3-matplotlib-data.noarch                                                           3.4.3-4.el9                          @epel              
python3-matplotlib-data-fonts.noarch                                                     3.4.3-4.el9                          @epel              
python3-matplotlib-tk.x86_64                                                             3.4.3-4.el9                          @epel              
python3-matplotlib-doc.x86_64                                                            3.4.3-4.el9                          epel               
python3-matplotlib-gtk3.x86_64                                                           3.4.3-4.el9                          epel               
python3-matplotlib-inline.noarch                                                         0.1.5-1.el9                          epel               
python3-matplotlib-qt5.x86_64                                                            3.4.3-4.el9                          epel               
python3-matplotlib-test-data.x86_64                                                      3.4.3-4.el9                          epel               
python3-matplotlib-wx.x86_64                                                             3.4.3-4.el9                          epel               
python3-pyqtgraph.noarch                                                                 0.13.7-4.el9                         epel               
qt5-qtgraphicaleffects.i686                                                              5.15.9-1.el9                         appstream          
qt5-qtgraphicaleffects.x86_64                                                            5.15.9-1.el9                         appstream          
qt5-qtgraphicaleffects-doc.noarch                                                        5.15.9-1.el9                         appstream          
qt6-qtgraphs.x86_64                                                                      6.6.2-1.el9                          epel               
qt6-qtgraphs-devel.x86_64                                                                6.6.2-1.el9                          epel               
qt6-qtgraphs-examples.x86_64                                                             6.6.2-1.el9                          epel               
[... cut ...]
qwt5-qt5-devel.x86_64                                                                    5.2.3a-12.20210522gita2b11e3.el9     epel               
qwt5-qt5-doc.noarch                                                                      5.2.3a-12.20210522gita2b11e3.el9     epel    
```

```
# pip list | grep -iE "matplotl|qtgraph|Qwt5"
matplotlib     
     
# python3
>>> import matplotlib
>>> matplotlib.__version__
'3.4.3'
```

This might be some artifact of me running this in a container though, so maybe it'd be better if some one who's running Rhel-9 "natively" tests this instead

Tanishq30052002, 2025-05-13:
We were supposed to install `pip install pyqtgraph`

We were able to view the plots then:
[image]

Also, an error is coming in the terminal despite the plot working.

[image]

danielcranston, 2025-05-15:
Thanks for testing @Tanishq30052002 ! So in summary there seems to be some incompatibility when the various python packages installed via dnf interplays, and `pip install pyqtgraph` at least gets things to a somewhat working state.

I'm happy to give you the assignee credit for this one, thanks again for the follow-up.

Tanishq30052002, 2025-05-15:
> Thanks for testing [@Tanishq30052002](https://github.com/Tanishq30052002) ! So in summary there seems to be some incompatibility when the various python packages installed via dnf interplays, and [...]

True