ros2/kilted_tutorial_party #460: Rviz

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Connext
- BuildType: Binary
- Chip: Amd64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros2/rviz)

## Checks
- [ ] **Check Rviz**

     Launch using the `rviz2` command.
     
     The rviz window should be generally usable and should be able to load some of the discovered visualizations.
     
     Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rviz2
       ```
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
       
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
       
  
    **Expect**
    
    1.
       You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
       
    1.
       You should see a laser scan emitting from the waving end of the dummy robot.
       
    1.
       Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

SANJAY-16012, 2025-05-06:
### Setup

    Rmw: Connext
    BuildType: Binary
    Chip: Amd64
    Os: Noble

I got this error message and the execution stops immediately..I was able to run demo publisher and subscriber successfully and i did it to ensure ros2 has installed correctly...

[image]

Yadunund, 2025-05-06:
Thanks for testing! This was fixed recently https://github.com/ros2/rviz/pull/1456 and the binaries with the fixes should be available https://github.com/ros/rosdistro/pull/45630

Could you try again after upgrading your binaries? 
```
sudo apt update && sudo apt upgrade -y
```

DasariChaitanya, 2025-05-06:
@Yadunund I have attempted updating the binaries, but it does not seem to locate the new release.

SANJAY-16012, 2025-05-07:
Even after updating binaries, it seems didn't resolve
[image]

DasariChaitanya, 2025-05-07:
@SANJAY-16012 
The new binary has been released which includes the updated rviz2 binary. [ROS 2 Kilted Kaiju - Beta 2](https://github.com/ros2/ros2/releases/tag/release-kilted-beta2-20250507) This new release has fixed the issue.

From my understanding `sudo apt upgrade` does not update the ros packages installed as part of the binary.

SANJAY-16012, 2025-05-08:
### Setup

    Rmw: Connext
    BuildType: Binary
    Chip: Amd64
    Os: Noble

### Checks

    ✅ Check Rviz
    ✅ TF 
    ✅ LaserScan

### Tried Commands:

        # User input in terminal 1
        ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py

        # User input in terminal 2
        rviz2

[image]

https://github.com/user-attachments/assets/040a855d-5624-45e0-aafd-745957a13daf

Thank you developers @DasariChaitanya @cottsay  for your support...

ahcorde, 2025-05-13:
Thank you for testing

closed 2025-05-13 by ahcorde