ros2/kilted_tutorial_party #461: Rviz

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Connext
- BuildType: Binary
- Chip: Amd64
- Os: Windows

## Links
- [Repo URL](https://github.com/ros2/rviz)

## Checks
- [ ] **Check Rviz**

     Launch using the `rviz2` command.
     
     The rviz window should be generally usable and should be able to load some of the discovered visualizations.
     
     Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rviz2
       ```
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
       
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
       
  
    **Expect**
    
    1.
       You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
       
    1.
       You should see a laser scan emitting from the waving end of the dummy robot.
       
    1.
       Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Poo-09, 2025-05-02:
Interested

Aniketkumarroy, 2025-05-10:
Tested on Windows 11 with two different versions of connext
# rti_connext_dds-5.3.1

[image]

[image]
# rti connext_dds-7.5.0

[image]

summary:- didn't work in both, in 5.3.1 it crashed and in 7.5.0 nothing opened and the command exited

zamantousif, 2025-05-15:
I tried with RTI 7.3.0 and `rviz2` does launch successfully.
[image]

The robot bringup example also seems to work.

[image]

> This is a sanity check to ensure that the GUI tools aren't fundamentally broken.
> 
> ## Setup
> * Rmw: Connext with `RTI 7.3.0`
> * BuildType: Binary
> * Chip: Amd64
> * Os: Windows
> 
> ## Checks
> * [ ] 
>   [x]  **Check Rviz**
>   Launch using the `rviz2` command.
>   The rviz window should be generally usable and should be able to load some of the discovered visualizations.
>   Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
>   details

@ahcorde Is there enough information to close this ticket now?

ahcorde, 2025-05-15:
Thank you for testing