ros2/kilted_tutorial_party #462: Rviz

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Connext
- BuildType: Binary
- Chip: Arm64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros2/rviz)

## Checks
- [ ] **Check Rviz**

     Launch using the `rviz2` command.
     
     The rviz window should be generally usable and should be able to load some of the discovered visualizations.
     
     Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rviz2
       ```
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
       
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
       
  
    **Expect**
    
    1.
       You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
       
    1.
       You should see a laser scan emitting from the waving end of the dummy robot.
       
    1.
       Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

aehils, 2025-05-07:
All tasks carried out successfully. RViz remained stable throughout and ran smoothly.

No outputs to provide.

aehils, 2025-05-07:
### Snippet of ROS doctor report

PLATFORM INFORMATION
system           : Linux
platform info    : Linux-6.8.0-59-generic-aarch64-with-glibc2.39
release          : 6.8.0-59-generic
processor        : aarch64

   RMW MIDDLEWARE
middleware name    : rmw_connextdds

   ROS 2 INFORMATION
distribution name      : kilted
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}

   TOPIC LIST
topic               : /clicked_point
publisher count     : 1
subscriber count    : 0
topic               : /goal_pose
publisher count     : 1
subscriber count    : 0
topic               : /initialpose
publisher count     : 1
subscriber count    : 0
topic               : /joint_states
publisher count     : 2
subscriber count    : 1
topic               : /map
publisher count     : 2
subscriber count    : 0
topic               : /robot_description
publisher count     : 2
subscriber count    : 0
topic               : /scan
publisher count     : 2
subscriber count    : 1
topic               : /tf
publisher count     : 2
subscriber count    : 1
topic               : /tf_static
publisher count     : 2
subscriber count    : 1

nobso, 2025-05-12:
Everything seems to be working as expected in the following env,
```
RMW Implementation: Connext
BuildType: Binary
OS: Ubuntu Noble
Chip: Arm64

```

Followed the installation instruction from https://docs.ros.org/en/kilted/Installation/Alternatives/Ubuntu-Install-Binary.html

ROS Distro: [ros2-kilted-beta2-20250507-linux-noble-arm64.tar.bz2](https://github.com/ros2/ros2/releases/download/release-kilted-beta2-20250507/ros2-kilted-beta2-20250507-linux-noble-arm64.tar.bz2)

```
sudo apt update && sudo apt install -q -y ros-kilted-rmw-connextdds
export RMW_IMPLEMENTATION=rmw_connextdds
```

> In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
>> You should see the dummy robot's kinematic tree waving back and forth in the rviz window.

[image]

> In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
>> You should see a laser scan emitting from the waving end of the dummy robot.

[image]

> Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.

[image]

[image]

ahcorde, 2025-05-12:
Thank you for testing

closed 2025-05-12 by ahcorde