ros2/kilted_tutorial_party #463: Rviz

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Connext
- BuildType: Debian
- Chip: Amd64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros2/rviz)

## Checks
- [ ] **Check Rviz**

     Launch using the `rviz2` command.
     
     The rviz window should be generally usable and should be able to load some of the discovered visualizations.
     
     Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rviz2
       ```
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
       
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
       
  
    **Expect**
    
    1.
       You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
       
    1.
       You should see a laser scan emitting from the waving end of the dummy robot.
       
    1.
       Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

virkhanna20, 2025-05-04:
I will do this.

zamantousif, 2025-05-09:
Tested with RMW connext and it worked fine. 

Screenshot for reference:
[image]

Please assign this to me and it could be closed as well! @mjcarroll @ahcorde :)

> ## Setup
> 
>     * Rmw: Connext
> 
>     * BuildType: Debian
> 
>     * Chip: Amd64
> 
>     * Os: Noble
> 
> ## Checks
> 
>       [x]  **Check Rviz**
>       Launch using the `rviz2` command.
>       The rviz window should be generally usable and should be able to load some of the discovered visualizations.
>       Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
>       details
>       **Try**
>       
>       1. # User input in terminal 1
>          ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
>       2. # User input in terminal 2
>          rviz2
>       3. In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
>       4. In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
>       
>       **Expect**
>       
>       1. You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
>       2. You should see a laser scan emitting from the waving end of the dummy robot.
>       3. Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.
> 
> 
> _You can find the code used to generate this test case [here](https://github.com/audrow/yatm)_

ahcorde, 2025-05-09:
Thank you for testing

closed 2025-05-09 by ahcorde