ros2/kilted_tutorial_party #464: Rviz

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Connext
- BuildType: Debian
- Chip: Arm64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros2/rviz)

## Checks
- [ ] **Check Rviz**

     Launch using the `rviz2` command.
     
     The rviz window should be generally usable and should be able to load some of the discovered visualizations.
     
     Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rviz2
       ```
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
       
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
       
  
    **Expect**
    
    1.
       You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
       
    1.
       You should see a laser scan emitting from the waving end of the dummy robot.
       
    1.
       Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

slowrunner, 2025-05-05:
rviz2 aborts with 

```
terminate called after throwing an instance of 'resource_retriever::Exception'
  what():  Error retrieving file [/opt/ros/kilted/share/rviz_default_plugins/icons/classes/TF.svg]:  Failed to open file
Aborted (core dumped)
```

looking at the directory - there is a file TF.png but not TF.svg
```
ubuntu@kilteddave:~/KiltedDave/ros2ws$ ls /opt/ros/kilted/share/rviz_default_plugins/icons/classes/
AccelStamped.png  FluidPressure.png       Interact.png     Path.png                PublishPoint.svg      RobotModel.png      Wrench.png
AxesColor.svg     FocusCamera.svg         LaserScan.png    PointCloud2.png         Range.png             Select.png          XYZ.svg
Axes.png          GridCells.png           Map.png          PointCloud.png          RelativeHumidity.png  SetGoal.png
AxisColor.svg     Grid.png                MarkerArray.png  PointStamped.png        RGB8.svg              SetInitialPose.png
Camera.png        Illuminance.png         Marker.png       Polygon.png             RGBF32.svg            src
DepthCloud.png    Image.png               Measure.svg      PoseArray.png           RobotJoint.png        Temperature.png
Effort.png        Intensity.svg           MoveCamera.png   Pose.png                RobotLinkNoGeom.png   TF.png
FlatColor.svg     InteractiveMarkers.png  Odometry.png     PoseWithCovariance.png  RobotLink.png         TwistStamped.png
ubuntu@kilteddave:~/KiltedDave/ros2ws$ 

```

zamantousif, 2025-05-09:
@slowrunner were you able to test this or are you still running into the above issue?

One thing you could try is to pull the latest docker image if you were using docker and try again. It works for me atleast.

[image]

slowrunner, 2025-05-09:
> [@slowrunner](https://github.com/slowrunner) were you able to test this or are you still running into the above issue?

Yes, It tested great after I did `sudo apt install ros-kilted-rqt*`, not sure why installing plugins alone didn't populate the menu, but after installing anything with rqt in the name, I was able to see and test arm64, debian, connextdds and fastdds successfully.

zamantousif, 2025-05-09:
Nice! 😀
Good to hear that you were able to test and that too on RPi 😆
I think this ticket could be closed.

ahcorde, 2025-05-09:
Thank you for testing

closed 2025-05-09 by ahcorde