ros2/kilted_tutorial_party #466: Rviz

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Cyclone
- BuildType: Binary
- Chip: Amd64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros2/rviz)

## Checks
- [ ] **Check Rviz**

     Launch using the `rviz2` command.
     
     The rviz window should be generally usable and should be able to load some of the discovered visualizations.
     
     Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rviz2
       ```
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
       
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
       
  
    **Expect**
    
    1.
       You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
       
    1.
       You should see a laser scan emitting from the waving end of the dummy robot.
       
    1.
       Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

seekbot, 2025-05-05:
Not sure if I am allowed to "claim" this issue ticket, but kinda this test using Docker image with logs/results [here](https://github.com/ros2/kilted_tutorial_party/issues/467#issuecomment-2848452216).

DasariChaitanya, 2025-05-07:
All checks passed. Rviz2 works smoothly without issues.
Used the latest binary [release](https://github.com/ros2/ros2/releases/tag/release-kilted-beta2-20250507)

> This is a sanity check to ensure that the GUI tools aren't fundamentally broken.
> 
> ## Setup
> * Rmw: Cyclone
> * BuildType: Binary
> * Chip: Amd64
> * Os: Noble
> 
> ## Links
> * [Repo URL](https://github.com/ros2/rviz)
> 
> ## Checks
 - [x] **Check Rviz**
>   Launch using the `rviz2` command.
>   The rviz window should be generally usable and should be able to load some of the discovered visualizations.
>   Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.

[image]

ahcorde, 2025-05-08:
Thank you for testing

closed 2025-05-08 by ahcorde