ros2/kilted_tutorial_party #467: Rviz

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Cyclone
- BuildType: Binary
- Chip: Amd64
- Os: Windows

## Links
- [Repo URL](https://github.com/ros2/rviz)

## Checks
- [ ] **Check Rviz**

     Launch using the `rviz2` command.
     
     The rviz window should be generally usable and should be able to load some of the discovered visualizations.
     
     Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rviz2
       ```
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
       
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
       
  
    **Expect**
    
    1.
       You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
       
    1.
       You should see a laser scan emitting from the waving end of the dummy robot.
       
    1.
       Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

seekbot, 2025-05-02:
I am interested.

seekbot, 2025-05-03:
Similar to other tickets, I am also encountering issues where I cannot launch RViz2. I only get the loading screen, and the following log:
`(pixi_ros2_rolling) C:\pixi_ws>ros2 run rviz2 rviz2`
`[INFO] [1746244289.874999600] [rviz2]: Stereo is NOT SUPPORTED`
`[INFO] [1746244289.875396700] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)`
`[INFO] [1746244289.914370600] [rviz2]: Stereo is NOT SUPPORTED`
`[ros2run]: Process exited with failure 3765269347`

seekbot, 2025-05-03:
Tried to do Rviz2 via Docker on Ubuntu Noble but encountered [error](https://github.com/ros2/rviz/issues/1454) as expected.

seekbot, 2025-05-05:
So a bit to add from this RViz [error](https://github.com/ros2/rviz/issues/1454):
> Tried Rviz2 with CycloneDDS via Docker on Ubuntu Noble but encountered [error](https://github.com/ros2/rviz/issues/1454) as expected.

I converted [TF png from RViz GitHub repo](https://github.com/ros2/rviz/blob/rolling/rviz_default_plugins/icons/classes/TF.png) to svg using online converter as a temp fix on my end, turns out the TF image isn't the only issue:

`terminate called after throwing an instance of 'resource_retriever::Exception'
  what():  Error retrieving file [/opt/ros/kilted/share/rviz_common/icons/classes/Identity.svg]: Failed to open file`

[image]

seekbot, 2025-05-10:
Tried it with [new release](https://github.com/ros2/ros2/releases), got it working now!

[image]

ahcorde, 2025-05-12:
Thank you for testing again!

closed 2025-05-12 by ahcorde