ros2/kilted_tutorial_party #468: Rviz

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Cyclone
- BuildType: Binary
- Chip: Arm64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros2/rviz)

## Checks
- [ ] **Check Rviz**

     Launch using the `rviz2` command.
     
     The rviz window should be generally usable and should be able to load some of the discovered visualizations.
     
     Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rviz2
       ```
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
       
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
       
  
    **Expect**
    
    1.
       You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
       
    1.
       You should see a laser scan emitting from the waving end of the dummy robot.
       
    1.
       Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

boby2965, 2025-05-05:
Hi, i have worked on this ticket 

My Setup:

```
ubuntu@kilted-desktop:~$ dpkg --print-architecture
arm64
ubuntu@kilted-desktop:~$ lsb_release -d
No LSB modules are available. Description:	Ubuntu 24.04.2 LTS
ubuntu@kilted-desktop:~$ export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp

ubuntu@kilted-desktop:~$ echo $RMW_IMPLEMENTATION
rmw_cyclonedds_cpp

```
i got an error while running the rviz2 command as mentioned below

In terminal 1:

```
ubuntu@kilted-desktop:~/Desktop$ ubuntu@kilted-desktop:~/Desktop$
ubuntu@kilted-desktop:~/Desktop$ ros2 launch dummy_robot_bringup dummy_robot_bri ngup_launch. py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2025-05-
05 - 16-47-45 - 009922-kilted-desktop-4726
[INFO] [Launch]: Default logging verbosity is set to INFO [INFO] [dummy_map_server-1]: process started with pid [4730]
[INFO] [robot_state_publisher-2]: process started with pid [4731]
LINFO] [dummy_joint_states-3]: process started with pid [4732]
[INFO] [dummy_ laser-41: process started with pid [4733]
[dummy_laser-4] [INFO] [1746443865.611717907] [dummy_laser]: angle inc: 0.004363/ [dummy_laser-4] [INFO] [1746443865.625563480] [dummy_laser]: scan size: 1081 [dummy_laser-4] [INFO] [1746443865.626261261] [dummy_laser]: scan time increment
0.000000
[robot_state_publisher-2] [INFO] [1746443865.656884585] [robot_state_publisher]:Robot initialized

```

In terminal 2:

```
ubuntu@kilted-desktop:-/Desktop$ rviz2
[INFO] (1746443879.263778991] [rvizz]: Stereo is NOT SUPPORTED [INFO] (1746443879.264150326] [rviz2]: OpenGl version: 3.1 (GLSL 1.4)
[INFO] [1746443879. 366983658] [rviz2]: Stereo is NOT SUPPORTED
terminate called after throwing an instance of 'resource_retriever: :Exception' what (): Error retrieving file [/hone/ubuntu/ros2_kilted/ros2-11nux/share/rviz
_default_plugins/icons/classes/Tf.svg]: Fatled to open file
Aborted (core dumped)

```

nobso, 2025-05-12:
Everything seems to be working as expected in the following env,
```
RMW Implementation: FASTDDS
BuildType: debian: pre-built debian packages from the apt repository
OS: Ubuntu Noble
Chip: Arm64
```
> You should see the dummy robot's kinematic tree waving back and forth in the rviz window.

[image]

> You should see a laser scan emitting from the waving end of the dummy robot.

[image]

> Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.

[image]

[image]

ahcorde, 2025-05-12:
Thank you for testing

closed 2025-05-12 by ahcorde