ros2/kilted_tutorial_party #47: Check ROS 2 interface CLI

tfoote, 2025-05-01:
## Setup
- Rmw: Zenoh
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check list**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 interface list
       ```
  
    **Expect**
    
    1.
       A large list of all messages, services, and actions is given.
    
  </details>
- [ ] **Check package**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 interface package turtlesim
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       turtlesim/srv/TeleportRelative
       turtlesim/srv/Kill
       turtlesim/msg/Pose
       turtlesim/srv/TeleportAbsolute
       turtlesim/srv/SetPen
       turtlesim/action/RotateAbsolute
       turtlesim/srv/Spawn
       turtlesim/msg/Color
       ```
  </details>
- [ ] **Check packages**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 interface packages
       ```
  
    **Expect**
    
    1.
       A list of all of the packages that define messages, services, or actions is given. An example is as follows.
       
    1.
       ```bash
       # stdout in terminal 1
       action_msgs
       action_tutorials_interfaces
       actionlib_msgs
       builtin_interfaces
       ...
       turtlesim
       unique_identifier_msgs
       visualization_msgs
       ```
  </details>
- [ ] **Check proto**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 interface proto std_msgs/msg/Header
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       "stamp:
         sec: 0
         nanosec: 0
       frame_id: ''
       "
       ```
  </details>
- [ ] **Check show**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 interface show std_msgs/msg/Header
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       # Standard metadata for higher-level stamped data types.
       # This is generally used to communicate timestamped data
       # in a particular coordinate frame.
       
       # Two-integer timestamp that is expressed as seconds and nanoseconds.
       builtin_interfaces/Time stamp
         int32 sec
         uint32 nanosec
       
       # Transform frame with which this data is associated.
       string frame_id
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

jncfa, 2025-05-02:
Tested with Ubuntu Docker Container for  amd64(under Rosetta 2) (ghcr.io/sloretz/ros-testing:kilted-desktop)
 
Check list:
[image]

[image]

Check package, The messages from turtlesim have been migrated to turtlesim_msgs, so I adjusted the command accordingly:

[image]

Check packages:

[image]

Check proto:

[image]

Check show:

[image]

ahcorde, 2025-05-02:
Thank you for testing