ros2/kilted_tutorial_party #472: Rviz

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Fastdds
- BuildType: Binary
- Chip: Amd64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros2/rviz)

## Checks
- [ ] **Check Rviz**

     Launch using the `rviz2` command.
     
     The rviz window should be generally usable and should be able to load some of the discovered visualizations.
     
     Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rviz2
       ```
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
       
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
       
  
    **Expect**
    
    1.
       You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
       
    1.
       You should see a laser scan emitting from the waving end of the dummy robot.
       
    1.
       Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

NickTziaros, 2025-05-02:
## Setup
Used the ros-testing:kilted-desktop image.

[image]

#### 🔴   `rviz2`

Runing the docker container as per the instructions 

`docker run --rm=true -ti ghcr.io/sloretz/ros-testing:kilted-desktop bash`

produced the output below.

```
qt.qpa.xcb: could not connect to display 
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

Aborted (core dumped)
```

Runing the docker container with 

`docker run -it \
  --env="DISPLAY=$DISPLAY" \
  --env="QT_X11_NO_MITSHM=1" \
  --volume="/tmp/.X11-unix:/tmp/.X11-unix:rw" \
  --gpus all \
  --privileged \
  --device /dev/dri \
  ghcr.io/sloretz/ros-testing:kilted-desktop`

and

`xhost +si:localuser:root` on host
works for the other Ros Distro containers. In this case i get :

```QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[INFO] [1746188298.006460932] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1746188298.006511449] [rviz2]: OpenGl version: 4.5 (GLSL 4.5)
[INFO] [1746188298.023746009] [rviz2]: Stereo is NOT SUPPORTED
terminate called after throwing an instance of 'resource_retriever::Exception'
  what():  Error retrieving file [/opt/ros/kilted/share/rviz_default_plugins/icons/classes/TF.svg]: Failed to open file
Aborted (core dumped)
```

[image]

### This might be just a docker permission thing though.

#### 🟢  `ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py`
```
[INFO] [launch]: All log files can be found below /root/.ros/log/2025-05-02-12-12-16-696947-41d6f35c8e35-91421
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dummy_map_server-1]: process started with pid [91424]
[INFO] [robot_state_publisher-2]: process started with pid [91425]
[INFO] [dummy_joint_states-3]: process started with pid [91426]
[INFO] [dummy_laser-4]: process started with pid [91427]
[dummy_laser-4] [INFO] [1746187936.743996563] [dummy_laser]: angle inc: 0.004363
[dummy_laser-4] [INFO] [1746187936.744060270] [dummy_laser]: scan size: 1081
[dummy_laser-4] [INFO] [1746187936.744068899] [dummy_laser]: scan time increment:       0.000000
[robot_state_publisher-2] [INFO] [1746187936.744031898] [robot_state_publisher]: Robot initialized

SANJAY-16012, 2025-05-06:
### Setup

    Rmw: Fastdds
    BuildType: Binary
    Chip: Amd64
    Os: Noble

I got this error message and the execution stops immediately..I was able to run demo publisher and subscriber successfully and i did it to ensure ros2 has installed correctly...

[image]

Yadunund, 2025-05-06:
Thanks for testing! This was fixed recently https://github.com/ros2/rviz/pull/1456 and the binares with the fixes should be available https://github.com/ros/rosdistro/pull/45630

Could you try again after upgrading your binaries? 
```
sudo apt update && sudo apt upgrade -y
```

DasariChaitanya, 2025-05-07:
All checks passed. Rviz2 works smoothly without issues.
Used the latest binary [release](https://github.com/ros2/ros2/releases/tag/release-kilted-beta2-20250507)

> This is a sanity check to ensure that the GUI tools aren't fundamentally broken.
> 
> ## Setup
> * Rmw: Fastdds
> * BuildType: Binary
> * Chip: Amd64
> * Os: Noble
> 
> ## Links
> * [Repo URL](https://github.com/ros2/rviz)
> 
> ## Checks
 - [x]  **Check Rviz**
>   Launch using the `rviz2` command.
>   The rviz window should be generally usable and should be able to load some of the discovered visualizations.
>   Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
>   

[image]

closed 2025-05-08 by ahcorde