ros2/kilted_tutorial_party #473: Rviz

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Fastdds
- BuildType: Binary
- Chip: Amd64
- Os: Windows

## Links
- [Repo URL](https://github.com/ros2/rviz)

## Checks
- [ ] **Check Rviz**

     Launch using the `rviz2` command.
     
     The rviz window should be generally usable and should be able to load some of the discovered visualizations.
     
     Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rviz2
       ```
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
       
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
       
  
    **Expect**
    
    1.
       You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
       
    1.
       You should see a laser scan emitting from the waving end of the dummy robot.
       
    1.
       Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Malay-dev, 2025-05-03:
### Environment Details
```
Build-Type: Binary
```

```
 PLATFORM INFORMATION
system           : Windows
platform info    : Windows-10-10.0.19045-SP0
release          : 10
processor        : Intel64 Family 6 Model 140 Stepping 1, GenuineIntel

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : kilted
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}

   TOPIC LIST
topic               : none
publisher count     : 0
subscriber count    : 0
```

### Commands Used
```
call C:\pixi_ws\ros2-windows\local_setup.bat
rviz2
```

### Observation
RViz window did not open
The RViz2 initial window with logo appears briefly and then exits immediately without showing any error message or logs in the terminal. No GUI remains, and no feedback is given unless manually run with verbose logging.

Recording

https://github.com/user-attachments/assets/f71bba79-58b8-4b09-9e5f-57a9931f0a4d

Yadunund, 2025-05-06:
Thanks for testing! This was fixed recently https://github.com/ros2/rviz/pull/1456 and the binaries with the fixes should be available https://github.com/ros/rosdistro/pull/45630

Could you try again after upgrading your binaries? 
```
sudo apt update && sudo apt upgrade -y
```

Yadunund, 2025-05-06:
@Malay-dev we're in the process of creating the Windows binaries with the fixes applied. Will update here once it's ready and you can download them again from https://github.com/ros2/ros2/releases.

Tanishq30052002, 2025-05-10:
> [@Malay-dev](https://github.com/Malay-dev) we're in the process of creating the Windows binaries with the fixes applied. Will update here once it's ready and you can download them again from https://github.com/ros2/ros2/releases.

@Yadunund I have testest it is working fine.

[image]

Tanishq30052002, 2025-05-10:
[image]

Malay-dev, 2025-05-10:
[image]

[image]

Yes, as tested by @Tanishq30052002, Rviz2 windows open successfully and work as expected. This confirms that the RViz2 GUI is properly integrated with the ROS 2 launch setup and can visualize robot state data as intended.

- [x] You should see the dummy robot's kinematic tree waving back and forth in the RViz window.
- [x] You should see a laser scan emitting from the waving end of the dummy robot.
- [x] Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.

Thanks @Yadunund and @Tanishq30052002

ahcorde, 2025-05-12:
Thank you for testing

closed 2025-05-12 by ahcorde