ros2/kilted_tutorial_party #474: Rviz

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Fastdds
- BuildType: Binary
- Chip: Arm64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros2/rviz)

## Checks
- [ ] **Check Rviz**

     Launch using the `rviz2` command.
     
     The rviz window should be generally usable and should be able to load some of the discovered visualizations.
     
     Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rviz2
       ```
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
       
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
       
  
    **Expect**
    
    1.
       You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
       
    1.
       You should see a laser scan emitting from the waving end of the dummy robot.
       
    1.
       Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

danielcranston, 2025-05-02:
## Setup

```bash
$ echo $RMW_IMPLEMENTATION
# not set, so default (Fastdds)
$ dpkg --print-architecture
arm64
$ lsb_release -d
No LSB modules are available.
Description:    Ubuntu 24.04.2 LTS
```

All terminals below first source the binary archive:

```bash
$ source ~/ros2_kilted_binary_archive/ros2-linux/setup.bash
```

## Checks

- [ ] ~**Check Rviz**~

Ran into an issue bringing up rviz2.

```bash
# Terminal 1
$ ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```

```bash
# Terminal 2
$ rviz2
TU: error: ../src/freedreno/vulkan/tu_knl.cc:315: failed to open device /dev/dri/renderD128 (VK_ERROR_INCOMPATIBLE_DRIVER)
TU: error: ../src/freedreno/vulkan/tu_knl.cc:315: failed to open device /dev/dri/renderD128 (VK_ERROR_INCOMPATIBLE_DRIVER)
[INFO] [1746169021.740427364] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1746169021.740545708] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[INFO] [1746169021.939447808] [rviz2]: Stereo is NOT SUPPORTED
terminate called after throwing an instance of 'resource_retriever::Exception'
  what():  Error retrieving file [/home/daniel/ros2_kilted_binary_archive/ros2-linux/share/rviz_default_plugins/icons/classes/TF.svg]: Failed to open file
Aborted (core dumped)
```

(Ignore the `VK_ERROR_INCOMPATIBLE_DRIVER` error, it's WSL related somehow. It appears with `rqt` as well, and that works fine).

The issue is that `TF.svg` is indeed missing. A `.png` exists, but no `.svg`:

```bash
$ cd /home/daniel/ros2_kilted_binary_archive/ros2-linux/share/rviz_default_plugins/icons/classes
$ ls TF.*
TF.png
```

What's strange to me though is that:
* The `TF.svg` does not exist in the source repo, neither in the [kilted](https://github.com/ros2/rviz/tree/kilted/rviz_default_plugins/icons/classes) nor [jazzy](https://github.com/ros2/rviz/tree/jazzy/rviz_default_plugins/icons/classes) branches.
* A quick peek into the `ros:jazzy` docker container (after `apt installing ros-jazzy-rviz2`) showed that `TF.svg` isn't present there either.

So I'm a bit perplexed.

danielcranston, 2025-05-08:
Seems fixed in the [beta 2 release](https://github.com/ros2/ros2/releases/tag/release-kilted-beta2-20250507) archive!

```
$ source ~/ros2_kilted_binary_archive_20250507/ros2-linux/setup.bash
$ rviz2
```

```
$ source ~/ros2_kilted_binary_archive_20250507/ros2-linux/setup.bash
$ ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```

[image]

ahcorde, 2025-05-08:
Thank you for testing

closed 2025-05-08 by ahcorde