ros2/kilted_tutorial_party #475: Rviz

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Chip: Amd64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros2/rviz)

## Checks
- [ ] **Check Rviz**

     Launch using the `rviz2` command.
     
     The rviz window should be generally usable and should be able to load some of the discovered visualizations.
     
     Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rviz2
       ```
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
       
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
       
  
    **Expect**
    
    1.
       You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
       
    1.
       You should see a laser scan emitting from the waving end of the dummy robot.
       
    1.
       Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

4-legends, 2025-05-02:
Testing this ticket

4-legends, 2025-05-02:
I am running into issues running rviz2

`legends@4legends-ASUS:~$ rviz2
[INFO] [1746165257.650547707] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1746165257.650624362] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[INFO] [1746165257.668592274] [rviz2]: Stereo is NOT SUPPORTED
terminate called after throwing an instance of 'resource_retriever::Exception'
  what():  Error retrieving file [/opt/ros/kilted/share/rviz_default_plugins/icons/classes/TF.svg]: Failed to open file
Aborted (core dumped)
`

I checked filesystem and there is no 'TF.svg'

[image]

4-legends, 2025-05-02:
Testing Video link: https://drive.google.com/file/d/1AahQBdhM9M4A-SE9_pby-NwG6UVbrY9d/view?usp=sharing

Logs:

`legends@4legends-ASUS:~$ rviz2
[INFO] [1746168901.305000420] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1746168901.305043427] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[INFO] [1746168901.318887258] [rviz2]: Stereo is NOT SUPPORTED
[ERROR] [1746168901.360817427] [rviz2]: Could not load display config: Error retrieving file [/opt/ros/kilted/share/rviz_common/icons/classes/Tool%20Properties.svg]: Failed to open file
[INFO] [1746169041.972155382] [rviz2]: Trying to create a map of size 100 x 100 using 1 swatches
[ERROR] [1746169042.011264405] [rviz2]: rviz/glsl120/indexed_8bit_image.vert
rviz/glsl120/indexed_8bit_image.frag
 GLSL link result : 
active samplers with a different type refer to the same texture image unit
[WARN] [1746169062.419486938] [rviz]: Joint 'single_rrbot_joint1' has no <limit> tag in URDF. Effort plugin needs to know the effort limit to determine the size of the corresponding visual marker. Effort display for this joint will be inhibited.
[WARN] [1746169062.419585151] [rviz]: Joint 'single_rrbot_joint2' has no <limit> tag in URDF. Effort plugin needs to know the effort limit to determine the size of the corresponding visual marker. Effort display for this joint will be inhibited.
`

`legends@4legends-ASUS:~$ ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
[INFO] [launch]: All log files can be found below /home/legends/.ros/log/2025-05-01-23-55-12-897587-4legends-ASUS-84859
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dummy_map_server-1]: process started with pid [84862]
[INFO] [robot_state_publisher-2]: process started with pid [84863]
[INFO] [dummy_joint_states-3]: process started with pid [84864]
[INFO] [dummy_laser-4]: process started with pid [84865]
[dummy_laser-4] [INFO] [1746168912.961202199] [dummy_laser]: angle inc:	0.004363
[dummy_laser-4] [INFO] [1746168912.961248136] [dummy_laser]: scan size:	1081
[dummy_laser-4] [INFO] [1746168912.961253065] [dummy_laser]: scan time increment: 	0.000000
[robot_state_publisher-2] [INFO] [1746168912.970119821] [robot_state_publisher]: Robot initialized
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[dummy_joint_states-3] [INFO] [1746169149.782413245] [rclcpp]: signal_handler(signum=2)
[dummy_laser-4] [INFO] [1746169149.782491090] [rclcpp]: signal_handler(signum=2)
[robot_state_publisher-2] [INFO] [1746169149.782491822] [rclcpp]: signal_handler(signum=2)
[dummy_map_server-1] [INFO] [1746169149.782451573] [rclcpp]: signal_handler(signum=2)
[INFO] [dummy_joint_states-3]: process has finished cleanly [pid 84864]
[INFO] [dummy_laser-4]: process has finished cleanly [pid 84865]
[INFO] [dummy_map_server-1]: process has finished cleanly [pid 84862]
[INFO] [robot_state_publisher-2]: process has finished cleanly [pid 84863]
`

Ran into rviz2 startup issues but after that it was pretty smooth

zamantousif, 2025-05-09:
The issues pointed out earlier with missing icons could be fixed by pulling the latest docker image to test this ticket. So I did not run int these issues atleast when testing with the Kilted testing party docker image.

I verified the use of rviz2 with the same setup above in ticket #268 , I suppose both the tickets could be closed now? @mjcarroll @ahcorde 

Here is a screenshot for reference:
[image]

ahcorde, 2025-05-09:
Thank you for testing

closed 2025-05-09 by ahcorde

4-legends, 2025-05-09:
@ahcorde 
I just saw that ticket was assigned to someone else. However, I did find the issue, fixed it and uploaded the results.

This message is not to get assigned to the issue but just to make aware that people are posting results again on already tested ticket to get assigned to more tickets. This happened multiple times with me.

ahcorde, 2025-05-09:
sorry @4-legends ping me there I will fix it

zamantousif, 2025-05-09:
> @ahcorde
> I just saw that ticket was assigned to someone else. However, I did find the issue, fixed it and uploaded the results.
>
> This message is not to get assigned to the issue but just to make [...]

@4-legends sorry for creating confusion and for my eagerness to post results to this ticket. 😅

I have noticed that multiple authors also collaborate on some tickets so there can be multiple assignees to a ticket which is why I was prompt to report results as I did spend some time this afternoon to get things working. My bad is I completely missed your drive link.

I did not intend to take your credit away. I will be more careful going ahead. 🙂

4-legends, 2025-05-09:
> [@4-legends](https://github.com/4-legends) sorry for creating confusion and for my eagerness to post results to this ticket. 😅
>
> I have noticed that multiple authors also collaborate on some tickets [...]

@zamantousif  I understand and my apologies if I came across rude but my intention was just to make group aware that there are some people intentionally want to create confusion just to get goodies.

I have no issues with multiple assignee for a ticket multiple sources verifying is better for community.

Regardless, thank you for testing as well🙏🏻 and no hard feelings from my end.