ros2/kilted_tutorial_party #477: Rviz

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Fastdds
- BuildType: Source
- Chip: Amd64
- Os: Rhel 9

## Links
- [Repo URL](https://github.com/ros2/rviz)

## Checks
- [ ] **Check Rviz**

     Launch using the `rviz2` command.
     
     The rviz window should be generally usable and should be able to load some of the discovered visualizations.
     
     Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rviz2
       ```
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
       
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
       
  
    **Expect**
    
    1.
       You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
       
    1.
       You should see a laser scan emitting from the waving end of the dummy robot.
       
    1.
       Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

danielcranston, 2025-05-08:
When installing from source in https://github.com/ros2/kilted_tutorial_party/issues/296#issuecomment-2860575648 I wasn't aware that https://github.com/ros2/rviz/pull/1455 was merged and backported to `kilted`. So I ended up building with that fix included, which was nice!

- [x]  **Check Rviz**

Some small caveats, though:

After struggling to get rviz2 to run inside the container I was using, I ended up with this:

```
$ docker run -it --rm -e DISPLAY=$DISPLAY -e XDG_RUNTIME_DIR=$XDG_RUNTIME_DIR -v /tmp/.X11-unix:/tmp/.X11-unix -v /run/user/1000:/run/user/1000  docker.io/library/almalinux:srcinstall 
```

I got the idea to add `XDG_RUNTIME_DIR` and mount `/run/user/1000` since rviz2 was giving me this error:
> error: XDG_RUNTIME_DIR is invalid or not set in the environment.

Even with the above, couldn't get rviz2 to run using GPU, so I ended up using

```
LIBGL_ALWAYS_SOFTWARE=1 rviz2
```

Else I would get the following:

```
# rviz2
QStandardPaths: runtime directory '/run/user/1000' is not owned by UID 0, but a directory permissions 0700 owned by UID 1000 GID 1000
MESA: error: Failed to query drm device.
glx: failed to create dri3 screen
failed to load driver: iris
MESA: error: Failed to query drm device.
glx: failed to create dri3 screen
failed to load driver: iris
MESA: error: Failed to query drm device.
glx: failed to create dri3 screen
failed to load driver: iris
```

It could just be some permissions issue in `/run/user/1000`, but since it worked with software rendering I decided not to pursue it further. Let me know if you want me to continue digging though!

Result:

[image]

ahcorde, 2025-05-13:
Thank you for testing

closed 2025-05-13 by ahcorde