ros2/kilted_tutorial_party #478: Rviz

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Fastdds
- BuildType: Source
- Chip: Amd64
- Os: Windows

## Links
- [Repo URL](https://github.com/ros2/rviz)

## Checks
- [ ] **Check Rviz**

     Launch using the `rviz2` command.
     
     The rviz window should be generally usable and should be able to load some of the discovered visualizations.
     
     Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rviz2
       ```
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
       
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
       
  
    **Expect**
    
    1.
       You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
       
    1.
       You should see a laser scan emitting from the waving end of the dummy robot.
       
    1.
       Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

yashphalle, 2025-05-03:
Hey @tfoote, I can work on this. Could you please assign this to me?

yashphalle, 2025-05-03:
# RViz2 Failing on Windows - ROS 2 Kilted-Pixi

## Environment
I have installed ROS 2 Kilted Kaiju following the official documentation (https://docs.ros.org/en/kilted/Installation.html) and successfully verified the installation with the talker/listener demo.

## Issue Description
When attempting to launch `rviz2`, no GUI window appears. Upon running `ros2 run rviz2 rviz2` explicitly, I encounter the following error:

```
[INFO] [1746293205.415834900] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1746293205.416093200] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[INFO] [1746293205.449921500] [rviz2]: Stereo is NOT SUPPORTED
[ros2run]: Process exited with failure 3221226505
```

## Troubleshooting Steps Taken
1. Updated graphics drivers to address potential compatibility issues
2. Verified other GUI ROS tools work correctly (e.g., `ros2 run rqt_graph rqt_graph.exe` launches successfully)

I am uncertain about the root cause of this issue. @tfoote  Could you please provide guidance on additional troubleshooting steps to ensure my setup is correct?

Tanishq30052002, 2025-05-08:
[image]
Working fine
@tfoote

ahcorde, 2025-05-09:
Thank you for testing

closed 2025-05-09 by ahcorde