ros2/kilted_tutorial_party #479: Rviz

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Zenoh
- BuildType: Binary
- Chip: Amd64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros2/rviz)

## Checks
- [ ] **Check Rviz**

     Launch using the `rviz2` command.
     
     The rviz window should be generally usable and should be able to load some of the discovered visualizations.
     
     Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rviz2
       ```
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
       
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
       
  
    **Expect**
    
    1.
       You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
       
    1.
       You should see a laser scan emitting from the waving end of the dummy robot.
       
    1.
       Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-02:
Hey, it didn't work. It crashed with the following error:

```
[INFO] [1746192315.393484359] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1746192315.393529231] [rviz2]: OpenGl version: 4.5 (GLSL 4.5)
[INFO] [1746192315.415765447] [rviz2]: Stereo is NOT SUPPORTED
terminate called after throwing an instance of 'resource_retriever::Exception'
  what():  Error retrieving file [/root/ros2_kilted/ros2-linux/share/rviz_default_plugins/icons/classes/TF.svg]: Failed to open file
Aborted (core dumped)

```

SANJAY-16012, 2025-05-08:
All checks are passed in testing Rviz2 and it works smoothly without any issues...

### Setup

    Rmw: Zenoh
    BuildType: Binary
    Chip: Amd64
    Os: Noble

### Checks

    ✅ Check Rviz
    ✅ TF
    ✅ LaserScan

### Commands

    # User input in terminal 1
    ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py

    # User input in terminal 2
    rviz2

[image]

https://github.com/user-attachments/assets/ca1553ee-3c8e-45da-b570-aaf53d2d1047

ahcorde, 2025-05-08:
Thank you for testing