ros2/kilted_tutorial_party #480: Rviz

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Zenoh
- BuildType: Binary
- Chip: Amd64
- Os: Windows

## Links
- [Repo URL](https://github.com/ros2/rviz)

## Checks
- [ ] **Check Rviz**

     Launch using the `rviz2` command.
     
     The rviz window should be generally usable and should be able to load some of the discovered visualizations.
     
     Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rviz2
       ```
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
       
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
       
  
    **Expect**
    
    1.
       You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
       
    1.
       You should see a laser scan emitting from the waving end of the dummy robot.
       
    1.
       Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

david-dorf, 2025-05-01:
Interested in this

cottsay, 2025-05-01:
Feel free to ping back here when you've wrapped up [the other ticket(s) assigned to you](https://github.com/ros2/kilted_tutorial_party/issues?q=is%3Aissue%20state%3Aopen%20assignee%3Adavid-dorf).

david-dorf, 2025-05-01:
Fails immediately (Windows AMD64). Let me know if there are debugging steps I can take.

https://github.com/user-attachments/assets/86b4283c-7b30-4a6f-ae8e-506d86c9ae5d

david-dorf, 2025-05-01:
Worth noting that `rviz2` by itself similarly shows that box, then closes.

Poo-09, 2025-05-02:
Interested in this

KajalAw, 2025-05-09:
Hi, I am testing this now. Will update the results shortly.

KajalAw, 2025-05-10:
[image]

Tested this out on Windows 10, AMD64, Build Type: Binary, rmw_zenoh_cpp

Successfully launched dummy_robot_bringup and rviz2

ahcorde, 2025-05-12:
Thank you for testing

closed 2025-05-12 by ahcorde