ros2/kilted_tutorial_party #482: Rviz

tfoote, 2025-05-01:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Setup
- Rmw: Zenoh
- BuildType: Debian
- Chip: Amd64
- Os: Noble

## Links
- [Repo URL](https://github.com/ros2/rviz)

## Checks
- [ ] **Check Rviz**

     Launch using the `rviz2` command.
     
     The rviz window should be generally usable and should be able to load some of the discovered visualizations.
     
     Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.
     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       rviz2
       ```
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
       
    1.
       In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
       
  
    **Expect**
    
    1.
       You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
       
    1.
       You should see a laser scan emitting from the waving end of the dummy robot.
       
    1.
       Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

avanmalleghem, 2025-05-02:
Interested in !

4-legends, 2025-05-04:
Testing this ticket

4-legends, 2025-05-04:
- Setup
```
legends@4legends-ASUS:~/ros2_ws$ printenv | grep ROS
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_DOMAIN_ID=0
ROS_AUTOMATIC_DISCOVERY_RANGE=SUBNET
ROS_DISTRO=kilted

legends@4legends-ASUS:~/ros2_ws$ printenv | grep RMW
RMW_IMPLEMENTATION=rmw_zenoh_cpp
``` 

- Issues
No other issues expect the one mentioned in #475 

- [x] Check Rviz

- Logs
```
legends@4legends-ASUS:~/ros2_ws$ ros2 run rmw_zenoh_cpp rmw_zenohd 
2025-05-04T23:47:03.199390Z  INFO ThreadId(02) zenoh::net::runtime: Using ZID: eabfc6e7aa67ee130ee0bda14fd6f6be
2025-05-04T23:47:03.199863Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/[2601:646:8f00:6450::184]:7447
2025-05-04T23:47:03.199873Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/[2601:646:8f00:6450:1c90:435b:551d:a0fa]:7447
2025-05-04T23:47:03.199875Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/[2601:646:8f00:6450:4f5f:f77b:4379:fc65]:7447
2025-05-04T23:47:03.199878Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/[fe80::637:667f:6818:5458]:7447
2025-05-04T23:47:03.199880Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/10.0.0.100:7447
Started Zenoh router with id eabfc6e7aa67ee130ee0bda14fd6f6be
^C[ros2run]: Received signal:  Interrupt
2025-05-04T23:50:07.513823Z  INFO ThreadId(02) zenoh::api::session: close session zid=eabfc6e7aa67ee130ee0bda14fd6f6be
``` 

```
legends@4legends-ASUS:~$ ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
[INFO] [launch]: All log files can be found below /home/legends/.ros/log/2025-05-04-16-47-12-368100-4legends-ASUS-41196
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dummy_map_server-1]: process started with pid [41199]
[INFO] [robot_state_publisher-2]: process started with pid [41200]
[INFO] [dummy_joint_states-3]: process started with pid [41201]
[INFO] [dummy_laser-4]: process started with pid [41202]
[dummy_laser-4] [INFO] [1746402432.481039909] [dummy_laser]: angle inc:	0.004363
[dummy_laser-4] [INFO] [1746402432.481151219] [dummy_laser]: scan size:	1081
[dummy_laser-4] [INFO] [1746402432.481168039] [dummy_laser]: scan time increment: 	0.000000
[robot_state_publisher-2] [INFO] [1746402432.494457427] [robot_state_publisher]: Robot initialized
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[dummy_laser-4] [INFO] [1746402605.651353189] [rclcpp]: signal_handler(signum=2)
[dummy_joint_states-3] [INFO] [1746402605.651389524] [rclcpp]: signal_handler(signum=2)
[robot_state_publisher-2] [INFO] [1746402605.651433247] [rclcpp]: signal_handler(signum=2)
[dummy_map_server-1] [INFO] [1746402605.651436809] [rclcpp]: signal_handler(signum=2)
[INFO] [dummy_joint_states-3]: process has finished cleanly [pid 41201]
[INFO] [dummy_laser-4]: process has finished cleanly [pid 41202]
[INFO] [dummy_map_server-1]: process has finished cleanly [pid 41199]
[INFO] [robot_state_publisher-2]: process has finished cleanly [pid 41200]
``` 

```
legends@4legends-ASUS:~$ rviz2
[INFO] [1746402438.256553794] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1746402438.256621952] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[INFO] [1746402438.278282356] [rviz2]: Stereo is NOT SUPPORTED
[ERROR] [1746402438.368484776] [rviz2]: Could not load display config: Error retrieving file [/opt/ros/kilted/share/rviz_common/icons/classes/Tool%20Properties.svg]: Failed to open file
``` 

https://github.com/user-attachments/assets/1663f61f-25a8-40fb-9d56-a73d0f6f3309

closed 2025-05-09 by mjcarroll