ros2/kilted_tutorial_party #485: RViz Custom Display

tfoote, 2025-05-01:
Check the documentation for the 'RViz Custom Display' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [RViz Custom Display page](https://docs.ros.org/en/kilted/Tutorials/Intermediate/RViz/RViz-Custom-Display/RViz-Custom-Display.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Tanishq30052002, 2025-05-03:
Hi 
I am testing this.

Tanishq30052002, 2025-05-08:
[image]

Will open a PR in the tutorial to update this.

Tanishq30052002, 2025-05-08:
needs to update it in the ros2 doc also
@ahcorde

Tanishq30052002, 2025-05-08:
Error expected when not wrap header:
```
[rviz2]: PluginlibFactory: The plugin for class 'rviz_plugin_tutorial::PointDisplay' failed to load. Error: Failed to load library /home/ros/ros2_ws/install/rviz_plugin_tutorial/lib/libpoint_display.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library LoadLibrary error: /home/ros/ros2_ws/install/rviz_plugin_tutorial/lib/libpoint_display.so: undefined symbol: _ZTVN20rviz_plugin_tutorial12PointDisplayE, at /tmp/binarydeb/ros-foxy-rcutils-1.1.4/src/shared_library.c:84
```

Error received:
```
[ERROR] [1746734178.883047840] [rviz2]: PluginlibFactory: The plugin for class 'Point2D' failed to load. Error: Failed to load library /root/ros2_ws/install/rviz_plugin_tutorial/lib/libpoint_display.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: /root/ros2_ws/install/rviz_plugin_tutorial/lib/libpoint_display.so: undefined symbol: _ZTVN20rviz_plugin_tutorial12PointDisplayE, at ./src/shared_library.c:96
```

Doesnt match

Tanishq30052002, 2025-05-08:
[image]

Even though no topic is subscribed the blue box is visible

Tanishq30052002, 2025-05-09:
> [image]
> 
> Will open a PR in the tutorial to update this. [@ahcorde](https://github.com/ahcorde)

Updated the PR

Tanishq30052002, 2025-05-09:
@ahcorde respected PR's have been created, please have a look.
Thanks

ahcorde, 2025-05-13:
Thank you for testing and openning the related PRs

closed 2025-05-13 by ahcorde

Tanishq30052002, 2025-05-13:
@ahcorde can you assign the issue to me?

ahcorde, 2025-05-13:
> [@ahcorde](https://github.com/ahcorde) can you assign the issue to me?

Sure, done!