ros2/kilted_tutorial_party #487: RViz User Guide

tfoote, 2025-05-01:
Check the documentation for the 'RViz User Guide' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [RViz User Guide page](https://docs.ros.org/en/kilted/Tutorials/Intermediate/RViz/RViz-User-Guide/RViz-User-Guide.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

leander-dsouza, 2025-05-02:
Hello @tfoote , I cannot seem to launch RViz.

## My Setup

* **RMW - FastDDS**

  I used the following command to set the rmw configuration type:
   
   ```bash
  export RMW_IMPLEMENTATION=rmw_fastrtps_cpp
  ```

* BuildType: Debian
* OS: Ubuntu 24.04 - Noble Numbat
* Chip: `x86_64`

  I used the following command to set the chip type:
   
   ```bash
  uname -m 
  ```

## Installation

I have used the binary installation for [Ubuntu (deb packages)](https://docs.ros.org/en/kilted/Installation/Ubuntu-Install-Debs.html)

## Issue - RViz does not launch due to missing asset

* Error message output:

   ```bash
   $ ros2 run rviz rviz

   [INFO] [1746210860.225618631] [rviz2]: Stereo is NOT SUPPORTED
   [INFO] [1746210860.225695311] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
   [INFO] [1746210860.283568587] [rviz2]: Stereo is NOT SUPPORTED
   terminate called after throwing an instance of 'resource_retriever::Exception'
     what():  Error retrieving file [/opt/ros/kilted/share/rviz_default_plugins/icons/classes/TF.svg]: Failed to open file
   [ros2run]: Aborted
   ```
* I think this is due that the TF asset is a `.png` file instead of `.svg`:

   ```bash
   $ ls /opt/ros/kilted/share/rviz_default_plugins/icons/classes/

   AccelStamped.png   GridCells.png           Marker.png        Pose.png                RobotModel.png
   AxesColor.svg      Grid.png                Measure.svg       PoseWithCovariance.png  Select.png
   Axes.png           Illuminance.png         MoveCamera.png    PublishPoint.svg        SetGoal.png
   AxisColor.svg      Image.png               Odometry.png      Range.png               SetInitialPose.png
   Camera.png         Intensity.svg           Path.png          RelativeHumidity.png    src
   DepthCloud.png     InteractiveMarkers.png  PointCloud2.png   RGB8.svg                Temperature.png
   Effort.png         Interact.png            PointCloud.png    RGBF32.svg              TF.png
   FlatColor.svg      LaserScan.png           PointStamped.png  RobotJoint.png          TwistStamped.png
   FluidPressure.png  Map.png                 Polygon.png       RobotLinkNoGeom.png     Wrench.png
   FocusCamera.svg    MarkerArray.png         PoseArray.png     RobotLink.png           XYZ.svg
   ```

ahcorde, 2025-05-02:
I'm working on a fix

Tanishq30052002, 2025-05-08:
[image]
@ahcorde i have tested the functionality and it is working all as expected

ahcorde, 2025-05-08:
Thank you for testing

closed 2025-05-08 by ahcorde