ros2/kilted_tutorial_party #490: Sane development environment

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Test message generation using .msg, .srv, .action and .idl on top of the installed environment.**

     Generate messages from .msg, .srv, and .action files, which will automatically create their .idl files. Additionally, ensure that messages can be generated from .idl files alone.

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-04:
Hi, I am working on this

suchetanrs, 2025-05-05:
I built `nav2_msgs` (https://github.com/ros-navigation/navigation2). 

To the `CMakeLists.txt` of `nav2_msgs` I added a `msg/Field.idl` to enable message generation from `.idl` files. 

The idl file is below:

```
module nav2_msgs {
  module msg {
    struct Field {
      uint16 id;

      float costs;
    };
  };
};
```
- [x] Test message generation using .msg, .srv, .action and .idl on top of the installed environment.

The script to run the test is here https://github.com/suchetanrs/kilted_tutorial_party/tree/main/windows_scripts

The full log output:

```
**********************************************************************
** Visual Studio 2022 Developer Command Prompt v17.14.0-pre.5.0
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************

C:\Program Files\Microsoft Visual Studio\2022\Preview>cd C:\pixi_ws

C:\pixi_ws>pixi shell

(pixi_ros2_rolling) C:\pixi_ws>call C:\pixi_ws\ros2-windows\local_setup.bat
"[rti_connext_dds_cmake_module][warning] RTI Connext DDS environment script not found (\resource\scripts\rtisetenv_x64Win64VS2017.bat). RTI Connext DDS will not be available at runtime, unless you already configured PATH manually."

(pixi_ros2_rolling) C:\pixi_ws>ros2 doctor --report | grep middleware
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 162: UserWarning: Fail to call NetworkReport class functions.
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 146: UserWarning: Report entry point PackageReport fails to load: No module named 'rosdistro'
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 146: UserWarning: Report entry point PlatformReport fails to load: No module named 'rosdistro'
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 146: UserWarning: Report entry point RosdistroReport fails to load: No module named 'rosdistro'
middleware name    : rmw_fastrtps_cpp

(pixi_ros2_rolling) C:\pixi_ws>cd kilted_tutorial_party\windows_scripts\

(pixi_ros2_rolling) C:\pixi_ws\kilted_tutorial_party\windows_scripts>.\sane_dev_env.cmd

(pixi_ros2_rolling) C:\pixi_ws\kilted_tutorial_party\windows_scripts>cd \ros2_ws

(pixi_ros2_rolling) C:\ros2_ws>rm -rf build log install

(pixi_ros2_rolling) C:\ros2_ws>colcon build --merge-install --packages-up-to nav2_msgs
Starting >>> geographic_msgs
Starting >>> nav2_common
Finished <<< nav2_common [3.36s]
[Processing: geographic_msgs]
[Processing: geographic_msgs]
Finished <<< geographic_msgs [1min 7s]
Starting >>> nav2_msgs
[Processing: nav2_msgs]
Finished <<< nav2_msgs [33.3s]

Summary: 3 packages finished [1min 41s]

(pixi_ros2_rolling) C:\ros2_ws>source install\local_setup.bat
'source' is not recognized as an internal or external command,
operable program or batch file.

(pixi_ros2_rolling) C:\ros2_ws>echo "============================ros2 interface show nav2_msgs/msg/EdgeCost"
"============================ros2 interface show nav2_msgs/msg/EdgeCost"
[... 164 lines cut ...]
    },
    "referenced_type_descriptions": []
  },
  "type_hashes": [
    {
      "type_name": "nav2_msgs/msg/Field",
      "hash_string": "RIHS01_cefd2012d995dddf6717de1fe4c6c562c8b7c9fd8ff4db01470dfccf50ba862f"
    }
  ]
}
(pixi_ros2_rolling) C:\ros2_ws>echo "============================cat the Field.idl"
"============================cat the Field.idl"

(pixi_ros2_rolling) C:\ros2_ws>cat install\share\nav2_msgs\msg\Field.idl
module nav2_msgs {
  module msg {
    struct Field {
      uint16 id;

      float costs;
    };
  };
};
(pixi_ros2_rolling) C:\ros2_ws>
```

closed 2025-05-09 by mjcarroll