ros2/kilted_tutorial_party #492: Sane development environment

tfoote, 2025-05-01:
## Setup
- Rmw: Zenoh
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Test message generation using .msg, .srv, .action and .idl on top of the installed environment.**

     Generate messages from .msg, .srv, and .action files, which will automatically create their .idl files. Additionally, ensure that messages can be generated from .idl files alone.

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

khaledgabr77, 2025-05-02:
I'm testing this.

khaledgabr77, 2025-05-02:
I have completed the following `.msg`, `.srv`, `.action` and `.idl` tests and confirmed everything works as expected.

## Checks

* [x]  **Test message generation using .msg, .srv, .action and .idl on top of the installed environment.**   

[image]

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde