ros2/kilted_tutorial_party #494: Service executables in `demo_nodes_py`

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Add two ints**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_py add_two_ints_client
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py add_two_ints_server
       ```
  
    
  </details>
- [ ] **Add two ints async**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_py add_two_ints_client_async
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py add_two_ints_server
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-04:
Successfully completed all checks for Issue #494 on Windows 11 / amd64 / binary / CycloneDDS.

1.  **Synchronous Client Check:** Ran `add_two_ints_server` in T1 and `add_two_ints_client` in T2. Server received the request and the client printed the correct result (3).
2.  **Asynchronous Client Check:** Ran `add_two_ints_server` in T1 and `add_two_ints_client_async` in T2. Server received the request and the client printed the correct result (3).

Both Python client/server demos worked as expected.

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde