ros2/kilted_tutorial_party #496: Service executables in `demo_nodes_py`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Add two ints**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_py add_two_ints_client
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py add_two_ints_server
       ```
  
    
  </details>
- [ ] **Add two ints async**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_py add_two_ints_client_async
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py add_two_ints_server
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

JasperGrant, 2025-05-02:
Add two ints:
  
  Client:
  `[INFO] [1746198282.417935700] [add_two_ints_client]: Result of add_two_ints: 5`
  Server:
  `[INFO] [1746198282.322058900] [add_two_ints_server]: Incoming request
  a: 2 b: 3`

Add two ints async:

  Client:
  `[INFO] [1746198394.842987900] [add_two_ints_client]: Result of add_two_ints: 5`
  Server:
  `[INFO] [1746198394.607912300] [add_two_ints_server]: Incoming request
  a: 2 b: 3`

Both tests passed. Ready to be closed!

ahcorde, 2025-05-02:
thank you for testing