ros2/kilted_tutorial_party #497: Service executables in `demo_nodes_py`

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Add two ints**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_py add_two_ints_client
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py add_two_ints_server
       ```
  
    
  </details>
- [ ] **Add two ints async**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run demo_nodes_py add_two_ints_client_async
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 run demo_nodes_py add_two_ints_server
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Tenveis, 2025-05-02:
I am interested.

Tenveis, 2025-05-02:
Following are the results:

*    I was able to follow the documentation
*    The documentation was clear
*    I didn't find any errors in the documentation

- [x]  Add two ints
[image]

- [x]   Add two ints async
[image]

closed 2025-05-07 by fujitatomoya