ros2/kilted_tutorial_party #499: Service introspection cpp

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [demo_nodes_cpp](https://github.com/ros2/demos/tree/kilted/demo_nodes_cpp)

## Checks
- [ ] **Run the client and server**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp introspect_services_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic list --include-hidden-topics
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       [introspection_service-1] a: 2 b: 3
       [introspection_client-2] [INFO] [1680585324.964598437] [introspection_client]: Result of add_two_ints: 5
       ```
    1.
       ```bash
       # stdout in terminal 2
       /parameter_events
       /rosout
       ```
  </details>
- [ ] **Configure service to send introspection metadata**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp introspect_services_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 param set /introspection_service service_configure_introspection metadata
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic echo /add_two_ints/_service_event
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       [introspection_service-1] a: 2 b: 3
       [introspection_client-2] [INFO] [1680585324.964598437] [introspection_client]: Result of add_two_ints: 5
       ```
    1.
       ```bash
       # stdout in terminal 2
       info:
         event_type: 1
         stamp:
           sec: 1680586531
           nanosec: 726539263
         client_gid:
         - 83
         - 143
         - 248
         - 38
         - 47
         - 231
         - 185
         - 230
         - 98
         - 17
         - 45
         - 187
         - 128
         - 124
         - 253
         - 156
         sequence_number: 46
       request: []
       response: []
       ---
       info:
         event_type: 2
         stamp:
           sec: 1680586531
           nanosec: 726695003
         client_gid:
         - 83
         - 143
         - 248
         - 38
         - 47
         - 231
         - 185
         - 230
         - 98
         - 17
         - 45
         - 187
         - 128
         - 124
         - 253
         - 156
         sequence_number: 46
       request: []
       response: []
       ```
  </details>
- [ ] **Configure client to send introspection metadata**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp introspect_services_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 param set /introspection_client client_configure_introspection metadata
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic echo /add_two_ints/_service_event
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       [introspection_service-1] a: 2 b: 3
       [introspection_client-2] [INFO] [1680585324.964598437] [introspection_client]: Result of add_two_ints: 5
       ```
    1.
       ```bash
       # stdout in terminal 2
       info:
         event_type: 0
         stamp:
           sec: 1680586637
           nanosec: 158736272
         client_gid:
         - 1
         - 16
         - 34
         - 142
         - 114
         - 128
         - 207
         - 116
         - 98
         - 246
         - 204
         - 52
         - 0
         - 0
         - 0
         - 7
         sequence_number: 48
       request: []
       response: []
       ---
       info:
         event_type: 3
         stamp:
           sec: 1680586637
           nanosec: 159157853
         client_gid:
         - 1
         - 16
         - 34
         - 142
         - 114
         - 128
         - 207
         - 116
         - 98
         - 246
         - 204
         - 52
         - 0
         - 0
         - 0
         - 7
         sequence_number: 48
       request: []
       response: []
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Tanishq30052002, 2025-05-03:
Hi
@tfoote can i go with this

Innomer, 2025-05-04:
Please assign this issue to me. Thank you.

ahcorde, 2025-05-05:
can you include the ouput ?

Innomer, 2025-05-05:
## Checks
- [x] Run the client and server

[image]

[image]

- [x] Configure service to send introspection metadata

[image]

- [x] Configure client to send introspection metadata

[image]

## Result
- SUCCESS
- Works perfectly fine without any warnings or errors

ahcorde, 2025-05-05:
Thank you for testing!

closed 2025-05-05 by ahcorde