ros2/kilted_tutorial_party #50: Check ROS 2 launch CLI

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Checks
- [ ] **Check launch package**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp talker_listener_launch.py
       ```
  
    **Expect**
    
    1.
       You should see something like the following
       ```bash
       # stdout in terminal 1
       [INFO] [launch]: All log files can be found below /home/parallels/.ros/log/2022-03-21-17-38-54-737866-parallels-Parallels-Virtual-Platform-3717
       [INFO] [launch]: Default logging verbosity is set to INFO
       [INFO] [talker-1]: process started with pid [3722]
       [INFO] [listener-2]: process started with pid [3724]
       [talker-1] [INFO] [1647909535.937731548] [talker]: Publishing: 'Hello World: 1'
       [listener-2] [INFO] [1647909535.938337387] [listener]: I heard: [Hello World: 1]
       [talker-1] [INFO] [1647909536.937932068] [talker]: Publishing: 'Hello World: 2'
       [listener-2] [INFO] [1647909536.938125486] [listener]: I heard: [Hello World: 2]
       [talker-1] [INFO] [1647909537.936162331] [talker]: Publishing: 'Hello World: 3'
       [listener-2] [INFO] [1647909537.936366391] [listener]: I heard: [Hello World: 3]
       ...
       ```
  </details>
- [ ] **Check launch launch file directly**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Download a launch file, for example with `curl`.
       ```bash
       # User input in terminal 1
       curl https://raw.githubusercontent.com/ros2/demos/master/demo_nodes_cpp/launch/topics/talker_listener_launch.py > /tmp/talker_listener_launch.py
       ```
    1.
       ```bash
       # User input in terminal 1
       ros2 launch /tmp/talker_listener_launch.py
       ```
  
    **Expect**
    
    1.
       You should see something like the following
       ```bash
       # stdout in terminal 1
       [INFO] [launch]: All log files can be found below /home/parallels/.ros/log/2022-03-21-17-38-54-737866-parallels-Parallels-Virtual-Platform-3717
       [INFO] [launch]: Default logging verbosity is set to INFO
       [INFO] [talker-1]: process started with pid [3722]
       [INFO] [listener-2]: process started with pid [3724]
       [talker-1] [INFO] [1647909535.937731548] [talker]: Publishing: 'Hello World: 1'
       [listener-2] [INFO] [1647909535.938337387] [listener]: I heard: [Hello World: 1]
       [talker-1] [INFO] [1647909536.937932068] [talker]: Publishing: 'Hello World: 2'
       [listener-2] [INFO] [1647909536.938125486] [listener]: I heard: [Hello World: 2]
       [talker-1] [INFO] [1647909537.936162331] [talker]: Publishing: 'Hello World: 3'
       [listener-2] [INFO] [1647909537.936366391] [listener]: I heard: [Hello World: 3]
       ...
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

JasperGrant, 2025-05-02:
Just checking off a few small ones here. Also stealing this one!

Check Launch Package:
```
[INFO] [launch]: All log files can be found below C:\Users\jaspe\.ros\log\2025-05-02-13-59-14-887234-MSI-4716
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [talker.EXE-1]: process started with pid [14032]
[INFO] [listener.EXE-2]: process started with pid [6344]
[talker.EXE-1] [INFO] [1746205156.134980600] [talker]: Publishing: 'Hello World: 1'
[listener.EXE-2] [INFO] [1746205156.135720200] [listener]: I heard: [Hello World: 1]
[talker.EXE-1] [INFO] [1746205157.138523500] [talker]: Publishing: 'Hello World: 2'
[listener.EXE-2] [INFO] [1746205157.138862900] [listener]: I heard: [Hello World: 2]
[talker.EXE-1] [INFO] [1746205158.133919600] [talker]: Publishing: 'Hello World: 3'
[listener.EXE-2] [INFO] [1746205158.134121600] [listener]: I heard: [Hello World: 3]`
```

Check Launch File Directly:
```
[INFO] [launch]: All log files can be found below C:\Users\jaspe\.ros\log\2025-05-02-14-29-45-614753-MSI-9152
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [talker.EXE-1]: process started with pid [16516]
[INFO] [listener.EXE-2]: process started with pid [1824]
[talker.EXE-1] [INFO] [1746206986.868396200] [talker]: Publishing: 'Hello World: 1'
[talker.EXE-1] [INFO] [1746206987.865942800] [talker]: Publishing: 'Hello World: 2'
[listener.EXE-2] [INFO] [1746206987.866324800] [listener]: I heard: [Hello World: 2]
[talker.EXE-1] [INFO] [1746206988.866760600] [talker]: Publishing: 'Hello World: 3'
[listener.EXE-2] [INFO] [1746206988.867060000] [listener]: I heard: [Hello World: 3]
```

Both tests passed. Ready to be closed!

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde