ros2/kilted_tutorial_party #503: Service introspection cpp

tfoote, 2025-05-01:
## Setup
- Rmw: Zenoh
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [demo_nodes_cpp](https://github.com/ros2/demos/tree/kilted/demo_nodes_cpp)

## Checks
- [ ] **Run the client and server**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp introspect_services_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic list --include-hidden-topics
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       [introspection_service-1] a: 2 b: 3
       [introspection_client-2] [INFO] [1680585324.964598437] [introspection_client]: Result of add_two_ints: 5
       ```
    1.
       ```bash
       # stdout in terminal 2
       /parameter_events
       /rosout
       ```
  </details>
- [ ] **Configure service to send introspection metadata**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp introspect_services_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 param set /introspection_service service_configure_introspection metadata
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic echo /add_two_ints/_service_event
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       [introspection_service-1] a: 2 b: 3
       [introspection_client-2] [INFO] [1680585324.964598437] [introspection_client]: Result of add_two_ints: 5
       ```
    1.
       ```bash
       # stdout in terminal 2
       info:
         event_type: 1
         stamp:
           sec: 1680586531
           nanosec: 726539263
         client_gid:
         - 83
         - 143
         - 248
         - 38
         - 47
         - 231
         - 185
         - 230
         - 98
         - 17
         - 45
         - 187
         - 128
         - 124
         - 253
         - 156
         sequence_number: 46
       request: []
       response: []
       ---
       info:
         event_type: 2
         stamp:
           sec: 1680586531
           nanosec: 726695003
         client_gid:
         - 83
         - 143
         - 248
         - 38
         - 47
         - 231
         - 185
         - 230
         - 98
         - 17
         - 45
         - 187
         - 128
         - 124
         - 253
         - 156
         sequence_number: 46
       request: []
       response: []
       ```
  </details>
- [ ] **Configure client to send introspection metadata**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 launch demo_nodes_cpp introspect_services_launch.py
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 param set /introspection_client client_configure_introspection metadata
       ```
    1.
       ```bash
       # User input in terminal 2
       ros2 topic echo /add_two_ints/_service_event
       ```
  
    **Expect**
    
    1.
       ```bash
       # stdout in terminal 1
       [introspection_service-1] a: 2 b: 3
       [introspection_client-2] [INFO] [1680585324.964598437] [introspection_client]: Result of add_two_ints: 5
       ```
    1.
       ```bash
       # stdout in terminal 2
       info:
         event_type: 0
         stamp:
           sec: 1680586637
           nanosec: 158736272
         client_gid:
         - 1
         - 16
         - 34
         - 142
         - 114
         - 128
         - 207
         - 116
         - 98
         - 246
         - 204
         - 52
         - 0
         - 0
         - 0
         - 7
         sequence_number: 48
       request: []
       response: []
       ---
       info:
         event_type: 3
         stamp:
           sec: 1680586637
           nanosec: 159157853
         client_gid:
         - 1
         - 16
         - 34
         - 142
         - 114
         - 128
         - 207
         - 116
         - 98
         - 246
         - 204
         - 52
         - 0
         - 0
         - 0
         - 7
         sequence_number: 48
       request: []
       response: []
       ```
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Amronos, 2025-05-02:
```console
amronos@dev-pc:~$ ros2 run rmw_zenoh_cpp rmw_zenohd
2025-05-02T15:08:18.393061Z  INFO ThreadId(02) zenoh::net::runtime: Using ZID: 7d40955f6e0fe7addaea85aa2884eb85
2025-05-02T15:08:18.394492Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/[2401:4900:1c66:f8bf:6f74:211d:e60a:ad61]:7447
2025-05-02T15:08:18.394539Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/[2401:4900:1c66:f8bf:85:b5d9:b19b:b04c]:7447
2025-05-02T15:08:18.394557Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/[fe80::4041:301d:80a2:7e90]:7447
2025-05-02T15:08:18.394572Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/192.168.1.5:7447
Started Zenoh router with id 7d40955f6e0fe7addaea85aa2884eb85
```

- [x]  Run the client and server

```console
amronos@dev-pc:~$ export RMW_IMPLEMENTATION=rmw_zenoh_cpp
amronos@dev-pc:~$ ros2 launch demo_nodes_cpp introspect_services_launch.py
[INFO] [launch]: All log files can be found below /home/amronos/.ros/log/2025-05-02-20-38-57-776089-dev-pc-68055
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [introspection_service-1]: process started with pid [68059]
[INFO] [introspection_client-2]: process started with pid [68060]
[introspection_service-1] [INFO] [1746198538.484947126] [introspection_service]: Incoming request
[introspection_service-1] a: 2 b: 3
[introspection_client-2] [INFO] [1746198538.485367768] [introspection_client]: Result of add_two_ints: 5
[introspection_service-1] [INFO] [1746198538.985352343] [introspection_service]: Incoming request
[introspection_service-1] a: 2 b: 3
[introspection_client-2] [INFO] [1746198538.985744762] [introspection_client]: Result of add_two_ints: 5

amronos@dev-pc:~$ export RMW_IMPLEMENTATION=rmw_zenoh_cpp
amronos@dev-pc:~$ ros2 topic list --include-hidden-topics
/parameter_events
/rosout
```

- [x] Configure service to send introspection metadata

```console
amronos@dev-pc:~$ export RMW_IMPLEMENTATION=rmw_zenoh_cpp
amronos@dev-pc:~$ ros2 launch demo_nodes_cpp introspect_services_launch.py
[INFO] [launch]: All log files can be found below /home/amronos/.ros/log/2025-05-02-20-41-07-290243-dev-pc-68470
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [introspection_service-1]: process started with pid [68473]
[INFO] [introspection_client-2]: process started with pid [68474]
[introspection_service-1] [INFO] [1746198667.986026771] [introspection_service]: Incoming request
[introspection_service-1] a: 2 b: 3
[introspection_client-2] [INFO] [1746198667.986652248] [introspection_client]: Result of add_two_ints: 5
[introspection_service-1] [INFO] [1746198668.485687691] [introspection_service]: Incoming request
[introspection_service-1] a: 2 b: 3
[introspection_client-2] [INFO] [1746198668.485949162] [introspection_client]: Result of add_two_ints: 5

amronos@dev-pc:~$ export RMW_IMPLEMENTATION=rmw_zenoh_cpp
amronos@dev-pc:~$ ros2 param set /introspection_service service_configure_introspection metadata
Set parameter successful
amronos@dev-pc:~$ ros2 topic echo /add_two_ints/_service_event
info:
  event_type: 1
  stamp:
    sec: 1746198692
    nanosec: 486319644
  client_gid:
  - 114
  - 121
  - 205
  - 233
  - 165
  - 63
  - 128
  - 50
  - 34
  - 29
  - 64
  - 97
  - 247
  - 111
  - 240
  - 62
  sequence_number: 50
request: []
response: []
---
info:
  event_type: 2
  stamp:
    sec: 1746198692
    nanosec: 486588209
  client_gid:
  - 114
  - 121
  - 205
  - 233
  - 165
  - 63
  - 128
  - 50
  - 34
  - 29
  - 64
  - 97
  - 247
  - 111
  - 240
  - 62
  sequence_number: 50
request: []
response: []
---
info:
  event_type: 1
  stamp:
    sec: 1746198692
    nanosec: 986174634
  client_gid:
  - 114
  - 121
  - 205
  - 233
  - 165
  - 63
  - 128
  - 50
  - 34
  - 29
  - 64
  - 97
  - 247
  - 111
  - 240
  - 62
  sequence_number: 51
request: []
response: []
---
```

- [x] Configure client to send introspection metadata

```console
amronos@dev-pc:~$ export RMW_IMPLEMENTATION=rmw_zenoh_cpp
amronos@dev-pc:~$ ros2 launch demo_nodes_cpp introspect_services_launch.py
[INFO] [launch]: All log files can be found below /home/amronos/.ros/log/2025-05-02-21-57-52-443745-dev-pc-70694
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [introspection_service-1]: process started with pid [70697]
[INFO] [introspection_client-2]: process started with pid [70698]
[introspection_service-1] [INFO] [1746203273.078537029] [introspection_service]: Incoming request
[introspection_service-1] a: 2 b: 3
[introspection_client-2] [INFO] [1746203273.079253757] [introspection_client]: Result of add_two_ints: 5
[introspection_service-1] [INFO] [1746203273.577946556] [introspection_service]: Incoming request
[introspection_service-1] a: 2 b: 3
[introspection_client-2] [INFO] [1746203273.578134136] [introspection_client]: Result of add_two_ints: 5

amronos@dev-pc:~$ export RMW_IMPLEMENTATION=rmw_zenoh_cpp
amronos@dev-pc:~$ ros2 param set /introspection_client client_configure_introspection metadata
Set parameter successful
amronos@dev-pc:~$ ros2 topic echo /add_two_ints/_service_event
info:
  event_type: 0
  stamp:
    sec: 1746203283
    nanosec: 577663336
  client_gid:
  - 63
  - 247
  - 192
  - 212
  - 29
  - 226
  - 2
  - 149
  - 21
  - 245
  - 253
  - 83
  - 175
  - 32
  - 148
  - 167
  sequence_number: 22
request: []
response: []
---
info:
  event_type: 3
  stamp:
    sec: 1746203283
    nanosec: 578316665
  client_gid:
  - 63
  - 247
  - 192
  - 212
  - 29
  - 226
  - 2
  - 149
  - 21
  - 245
  - 253
  - 83
  - 175
  - 32
  - 148
  - 167
  sequence_number: 22
request: []
response: []
---
info:
  event_type: 0
  stamp:
    sec: 1746203284
    nanosec: 77638589
  client_gid:
  - 63
  - 247
  - 192
  - 212
  - 29
  - 226
  - 2
  - 149
  - 21
  - 245
  - 253
  - 83
  - 175
  - 32
  - 148
  - 167
  sequence_number: 23
request: []
response: []
---
```

ahcorde, 2025-05-02:
Thank you for testing