ros2/kilted_tutorial_party #504: Setting up efficient intra-process communication

tfoote, 2025-05-01:
Check the documentation for the 'Setting up efficient intra-process communication' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Setting up efficient intra-process communication page](https://docs.ros.org/en/kilted/Tutorials/Demos/Intra-Process-Communication.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

RishikesavanRamesh, 2025-05-02:
Hi, can I try this?

JatinVira, 2025-05-05:
I tested this out here are some snaps

[image]

[image]

- I was able to follow the documentation.
- The documentation seemed clear to me.
- The documentation didn't have any obvious errors.

Thanks

JatinVira, 2025-05-05:
Here's my setup

```
   PLATFORM INFORMATION
system           : Linux
processor        : x86_64

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : kilted
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}
```

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde