ros2/kilted_tutorial_party #505: Simulation Gazebo

tfoote, 2025-05-01:
Check the documentation for the 'Simulation Gazebo' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Simulation Gazebo page](https://docs.ros.org/en/kilted/Tutorials/Advanced/Simulators/Gazebo/Simulation-Gazebo.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

leander-dsouza, 2025-05-02:
The installation works for Gazebo Ionic.

## Installation Steps

* Install Gazebo Ionic:

   ```bash
   sudo apt-get install ros-${ROS_DISTRO}-ros-gz
   ```
* Run the Gazebo simulation:

   ```bash
   gz sim
   ```
   [image]

## Checks

- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**

## Errors
The links on the [Gazebo setup](https://docs.ros.org/en/kilted/Tutorials/Advanced/Simulators/Gazebo/Gazebo.html) redirect to harmonic instead of **ionic**.

j-rivero, 2025-05-05:
> The links on the [Gazebo setup](https://docs.ros.org/en/kilted/Tutorials/Advanced/Simulators/Gazebo/Gazebo.html) redirect to harmonic instead of **ionic**.

This is correct. We expect Gazebo Ionic to be part of ROS 2 Kilted instead of Harmonic.

Yadunund, 2025-05-05:
@leander-dsouza thanks for flagging this. I've opened https://github.com/ros2/ros2_documentation/pull/5549 to update the documentation on `rolling` which we will backport to `kilted.

Yadunund, 2025-05-06:
@leander-dsouza i'll close this ticket given the documentation will be updated once the PR is merged but please feel free to reopen if something else needs to be fixed as well.

closed 2025-05-06 by Yadunund